verilator/verilator #4007: CMake support doesn't work with generated code

Timmmm, 2023-03-07:
The CMake support you've added is great but unfortunately I think it is done in a way that makes it basically impossible to use with any generated code (which is very common in my experience).

For example, suppose you would normally do this:

```
g++ -shared foo.cpp -o libfoo.so
verilator ... main.cpp top.sv libfoo.so
```

Seems pretty reasonable, but it's more or less impossible with the current CMake support because the `g++` command will run at *build* time, and the `verilator` command will run at *configure* time (before `libfoo.so` exists).

Fundamentally, the `verilator` command should run at *build* time instead using `add_custom_command()`.

I had a look at the code and actually it looks like it does do that, but then there's also some other code that calls `execute_process()` too which is what causes the problem.

I would guess this is a [dynamic dependency](https://ninja-build.org/manual.html#ref_dyndep) issue.

Timmmm, 2023-03-08:
Actually my description was a bit off - linking with `libfoo.so` is fine, the issue is if you generate a `.sv` file. In our case I could simply move that generation to configure time too. It's not ideal but it does work:

```
set_property(DIRECTORY APPEND PROPERTY CMAKE_CONFIGURE_DEPENDS "${CMAKE_CURRENT_SOURCE_DIR}/input_file.foo")

execute_process(
    COMMAND "my_generator" -i "${CMAKE_CURRENT_SOURCE_DIR}/input_file.foo" -o "output_file.sv"
    COMMAND_ERROR_IS_FATAL ANY
)

...
```

wsnyder, 2023-03-16:
@codelec   for thoughts.

codelec, 2023-03-17:
looks similar to the issue here https://stackoverflow.com/questions/44499435/how-can-i-run-cmake-from-within-cmake

Timmmm, 2023-03-17:
I think the best way to fix this would be to add a single-output-file mode to Verilator (which would be quite nice anyway!). That way you don't need to actually run Verilator to know what the input files to the output binary target are.

However in my case generating the SV code at configure time is actually a reasonable solution so it's less of an issue for me than I thought. Feel free to close this if you like!

wsnyder, 2023-03-18:
Single file will not scale to reasonable sized designs, so I'd prefer not to go that way.

codelec, 2023-03-22:
an example use case found here
https://github.com/codelec/chisel-template/blob/ead66374a8bdf4fd1cbbdc984de89d3de0dfcbab/emulator/CMakeLists.txt
https://github.com/codelec/chisel-template/blob/ead66374a8bdf4fd1cbbdc984de89d3de0dfcbab/CMakeLists.txt

sivansh11, 2024-09-11:
hey, I have a similar usecase, I am generating multiple source files from a different script, what would be the recommended way to implement this in cmake so it is built properly ?