verilator/verilator #4009: Errors in compile stage for Verilator 5.008

eminakgun, 2023-03-08:
Trying to build verilator 5.008,

```
git clone https://github.com/verilator/verilator   # Only first time
unsetenv VERILATOR_ROOT
cd verilator
git pull
git checkout v{5.008}
autoconf
./configure
make
```
output,

```
Compile flags:  g++ -std=gnu++11 -Og -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include 
-MP -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" 
-DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/usr/local/share/verilator"                                          
g++ -std=gnu++11  -Og -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include -MP  
-Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" 
-DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../Verilator.cpp -o Verilator.o 

In file included from ../V3Number.h:24:0,
from ../V3Ast.h:28,                                                      
from ../V3Assert.h:23,  
from ../Verilator.cpp:19: 
../V3Hash.h:75:13: error: specialization of ‘template<class _Tp> struct std::hash’ in different namespace [-fpermissive] struct std::hash<V3Hash> {  
                                                                                                                                     ^                            
In file included from /usr/include/c++/4.8.2/bits/basic_string.h:3033:0,                                                                                           
from /usr/include/c++/4.8.2/string:52,                                                                                                            
from /usr/include/c++/4.8.2/bits/locale_classes.h:40,                                                                                             
from /usr/include/c++/4.8.2/bits/ios_base.h:41,                                                                                                   
from /usr/include/c++/4.8.2/ios:42,                                                                                                               
from /usr/include/c++/4.8.2/ostream:38,                                                                                                           
from /usr/include/c++/4.8.2/iostream:39,                                                                                                         
 from ../config_build.h:65,                                                                                                                        
 from ../V3Active.h:20,                                                                                                                            
 from ../Verilator.cpp:17:                                                                                                        
 /usr/include/c++/4.8.2/bits/functional_hash.h:58:12: error:   from definition of ‘template<class _Tp> struct std::hash’ [-fpermissive]                 
 struct hash;  
		^
		
In file included from ../V3Assert.h:23:0,                                                                                                                          
from ../Verilator.cpp:19:                                                                                                        
../V3Ast.h:2471:33: error: specialization of ‘template<class _Tp> struct std::hash’ in different namespace [-fpermissive]                          
struct std::hash<VNRef<T_Node>> final {                                                                                                                                           
								^                                                                                                                
								
In file included from /usr/include/c++/4.8.2/bits/basic_string.h:3033:0,                                                                                           
from /usr/include/c++/4.8.2/string:52,                                                                                                            
from /usr/include/c++/4.8.2/bits/locale_classes.h:40,                                                                                             
from /usr/include/c++/4.8.2/bits/ios_base.h:41,                                                                                                   
from /usr/include/c++/4.8.2/ios:42,                                                                                                               
from /usr/include/c++/4.8.2/ostream:38,                                                                                                           
from /usr/include/c++/4.8.2/iostream:39,                                                                                                          
from ../config_build.h:65,                                                                                                                        
from ../V3Active.h:20,                                                                                                                            
from ../Verilator.cpp:17:                                                                                                        
/usr/include/c++/4.8.2/bits/functional_hash.h:58:12: error:   from definition of ‘template<class _Tp> struct std::hash’ [-fpermissive]                 
struct hash; 
		^

In file included from ../V3Assert.h:23:0,                                                                                                                          
from ../Verilator.cpp:19:                                                                                                        
../V3Ast.h:2477:37: error: specialization of ‘template<class _Tp> struct std::equal_to’ in different namespace [-fpermissive]                      
struct std::equal_to<VNRef<T_Node>> final {                                                                                                                                           
									^                                                                                                            
									
In file included from /usr/include/c++/4.8.2/string:48:0,                                                                                                          

from /usr/include/c++/4.8.2/bits/locale_classes.h:40,                                                                                             
from /usr/include/c++/4.8.2/bits/ios_base.h:41,                                                                                                   
from /usr/include/c++/4.8.2/ios:42,                                                                                                               
from /usr/include/c++/4.8.2/ostream:38,                                                                                                           
from /usr/include/c++/4.8.2/iostream:39,                                                                                                          
from ../config_build.h:65,                                                                                                                        
from ../V3Active.h:20,                                                                                                                            
from ../Verilator.cpp:17:                                                                                                        
/usr/include/c++/4.8.2/bits/stl_function.h:204:12: error:   from definition of ‘template<class _Tp> struct std::equal_t ’ [-fpermissive]                                                                                                                                       
struct equal_to : public binary_function<_Tp, _Tp, bool>                                                                                                 
		^                                                       

```

Could it be due to g++ version?

```
g++ --version
g++ (GCC) 4.8.5 20150623 (Red Hat 4.8.5-44) 
```

wsnyder, 2023-03-09:
Yes, this is a known GCC bug, see if adding the -Fpermissive flag solves the issue, or use a newer OS/gcc, and/or edit the code appropriately.

wsnyder, 2023-03-09:
Closing as presumes answers it, feel free to follow up here.

closed 2023-03-09 by wsnyder