verilator/verilator #4013: Internal error when declaring parametric mailbox instance inside parametric module using type parameter

botello, 2023-03-09:
Internal Error reported when trying to verilate this example.

```verilog
// tbtop.sv
module dummy #(parameter type T=logic);
    mailbox #(T) mbox;
endmodule
module tbtop;
    typedef struct packed {
        logic a;
        logic b;
    } my_struct_t;
    dummy #(my_struct_t) u_dummy();
endmodule
```

Command
`verilator -cc --exe -x-assign fast -Wall --trace --assert --debug --main --timing tbtop.sv --top-module tbtop`

Version `Verilator 5.006 2023-01-22 rev v5.006`
System `Rocky Linux 8.7 (Green Obsidian) -- rhel/centos/fedora like`

Error message
```
...
- V3LinkDot.cpp:3401: linkDotPrimary: 
- V3LinkJump.cpp:322: linkJump: 
- V3Ast.cpp:1122:     Dumping obj_dir/Vtbtop_006_linkjump.tree
- V3LinkInc.cpp:292:  linkIncrements: 
- V3Ast.cpp:1122:     Dumping obj_dir/Vtbtop_007_linkinc.tree
- V3Param.cpp:1312:   param: 
- V3LinkDot.cpp:3407: linkDotParamed: 
- V3Ast.cpp:1122:     Dumping obj_dir/Vtbtop_009_linkdotparam.tree
%Error: Internal Error: example.sv:2:5: ../V3Broken.cpp:174: Broken link in node (or something without maybePointedTo): 'm_classp && !m_classp->brokeExists()' @ ../V3AstNodes.cpp:1497
                                      : ... In instance tbtop.u_dummy
-node: CLASSREFDTYPE 0x164ce70 <e819> {d2af} @dt=this@(w0)class:mailbox  mailbox cpkg=0x1623b80 -> CLASS 0x1623b80 <e1267> {e29af}  mailbox  L5 [1ps]
    2 |     mailbox #(T) mbox;
      |     ^~~~~~~
- V3Ast.cpp:1122:     Dumping obj_dir/Vtbtop_990_final.tree
- V3Broken.cpp:327:   Broken called under broken, skipping recursion.
- V3StatsReport.cpp:225:statsReport: 
%Error: Internal Error: Aborting since under --debug
...
```

wsnyder, 2023-03-10:
This seems to be related to class parameterization, the "remove classes that have void params" is removing something with a reference.

Perhaps you could look at making a pull request to fix the broken reference and/or @kbieganski can help as he did that part?

kbieganski, 2023-03-10:
Thanks for the minimal test case. The issue is indeed with class params (not `mailbox`), as this also fails:
```
// tbtop.sv
class foo #(type T);
endclass
module dummy #(parameter type T=logic);
    foo #(T) mbox;
endmodule
module tbtop;
    typedef struct packed {
        logic a;
        logic b;
    } my_struct_t;
    dummy #(my_struct_t) u_dummy();
endmodule
```
It does compile without `--debug`, so not a huge bug. The parametrized module indirection is key here: put the `mailbox` in a concrete module, or in a parametrized class and it works. (Also note you need `-Wno-DECLFILENAME` for this file)

I only worked on `mailbox`, and reviewed some class param code, but my knowledge is limited. Happy to help though. @RRozak any ideas?

RRozak, 2023-03-10:
The problem occurs, because after `linkDotParamed` step, there is an instance of `dummy` module with the default parameter value and it contains a reference to `foo` with the default parameter value too. But that class instance isn't created, because it is not needed and the default instance of `dummy` is removed one step later (in V3Dead).
So it is not a real problem and if you don't need to use `--debug` flag, you can ignore it. If you want to use that flag, you can add a use of module instance with the default parameter value (just put `dummy unused();` at the end of the top module).

wsnyder, 2023-03-10:
Note it can't be ignored, removing --debug just exchanges one internal error for another.

botello, 2023-03-13:
I tried both suggestions:
1. Run without `--debug` flag. It seems to be working, at least this simple example verilates and builds. Why is the result different wether this flag is present or not. I was assuming it to be identical but with added verbosity.
2. Declare a default module instance using default paramter -- no `#(T)` override. This also works at the expense of adding an additional component.

```verilog
// tbtop.sv
module dummy #(parameter type T=logic);
    mailbox #(T) mbox;
endmodule
module tbtop;
    typedef struct packed {
        logic a;
        logic b;
    } my_struct_t;
    dummy #(my_struct_t) u_dummy();

    dummy u_unused();  // <-- added as suggested (2). This should preserve the objects and type T paramter
endmodule
```

What is your recommendation?

kbieganski, 2023-03-13:
> 1. Why is the result different wether this flag is present or not

Ideally it shouldn't, but debug mode does some additional checks which slows down Verilator but is useful for development. This is simply a bug that slipped past these checks. At some point it will be fixed. If you're up to it, feel free to contribute, all help is appreciated.

wsnyder, 2023-03-13:
On my system without --debug it core dumps.  It totally depends on what the memory layout is on your system and how new/delete reclaim memory.

closed 2025-09-27 by wsnyder