verilator/verilator #4016: SVA: Support for default disable iff

svenka3, 2023-03-11:
What would you like added/supported?

Default disable construct for SVA.

I believe this is similar to #1482 though slightly different construct here.

```verilog
module m;
  bit clk, rst_n;

  default disable iff (!rst_n);
  default clocking dcb @(posedge clk);
  endclocking : dcb 
endmodule : m

```

Above code gives error as:
```log
%Error: af_vlt_sva_def_dis.sv:4:11: syntax error, unexpected disable, expecting clocking
    4 |   default disable iff (!rst_n);
      |           ^~~~~~~
```

May we assist you in trying to fix this in Verilator yourself?

Yes, but my exposure to Verilator source code (and parsers etc.) is minimal, so will attempt.

wsnyder, 2023-03-11:
There have been recent parser improvements, if you run on trunk it should give you a better UNSUPPORTED error rather than a syntax error.  Anyhow to support this, first please make a self checking test, then add a new e.g. AstDefaultIff node type that is parsed in verilog.y, and plumb it through to V3Assert, perhaps like AstClocking is (for default clocking) though clocking blocks are more complicated so this won't need to do as much.

wsnyder, 2023-03-11:
Note as opposed to your #4017 this should be straight forward.  #1482 to fix requires a lot of changes where clocks are evaluated, but for this issue alone as at present assertions are just converted to always statements, an "iff x" can just be made into an "if (x)" inside V3Assert where the assertion is converted to an always statement.

svenka3, 2023-03-12:
Thanks a lot for the advice. I glanced through V3Assert.cpp and need more time, I believe to see where you convert this to an _always_ 

Wondering if there is a way to write out the converted verilog code for debug purposes? Like other tools have a "tokens.v" generation

Also the verilog.y is to be hand edited? Sorry if this is a dumb question, have heard it gets generated in other places (I could have misunderstood)

wsnyder, 2023-03-12:
Yes verilog.y is by hand.  See the docs/internals, you can use -debug to see the AstTree at each stage.

closed 2024-11-27 by wsnyder