verilator/verilator #4017: SVA: Delay at start of property definition does not compile

svenka3, 2023-03-11:
Below code does not compile in Verilator (Dev). 

```verilog
module m;

  bit clk;
  bit penable;

  property p;        
    @(posedge clk)
    ##1 (penable) |=> !(penable);       
  endproperty : p

endmodule : m

```

LOG:
```log
%Error: af_vlt_sva_del_at_start.sv:8:5: syntax error, unexpected ##, expecting TYPE-IDENTIFIER
    8 |     ##1 (penable) |=> !(penable);
```

What 'verilator --version' are you using?  Did you try it with the git master version?

> Verilator 5.007 devel rev v5.006-7-g4a8cfe367

What OS and distribution are you using?
> WSL
May we assist you in trying to fix this in Verilator yourself?

> Yes, will attempt

wsnyder, 2023-03-11:
There's no modeling of delayed assertions presently, the first step is to make some good tests to understand the problem.  Implementing delayed assertions might be fairly complicated so I suggest posing here your understanding of the problem and suggested approach algorithmically.  After that we can get into where to put the code.

wsnyder, 2025-09-09:
See also #4263