verilator/verilator #4026: Surprisingly slow compile times

oharboe, 2023-03-15:
The Verilog example below, is surprisingly slow to compile.

Any idea on what is going on?

I think this can ultimately make g++ crash. I'm struggling with a crash in g++ on .cpp code generated by Verilator.

```
verilator -cc test/Example.sv
# this takes minutes seconds on my machine
make -f VExample.mk
```

https://groups.google.com/g/chisel-users/c/iFA-oTbuk4Y

https://scastie.scala-lang.org/nXXNW6C1R1GLZbq5YcmqHw

wsnyder, 2023-03-15:
Example.v takes 0.445 seconds on the current master, please try that.

oharboe, 2023-03-16:
g++ crashes when I use v5.006 on our proprietary project:

```
[error] g++: internal compiler error: Segmentation fault signal terminated program cc1plus
[error] Please submit a full bug report, with preprocessed source (by using -freport-bug).
[error] See <file:///usr/share/doc/gcc-12/README.Bugs> for instructions.
```

oharboe, 2023-03-16:
I tried clang on the file that g++ crashes on: after 18 hours, clang had not crashed, nor completed compilation on that file.

oharboe, 2023-03-16:
I solved the problem by breaking out some of offending Verilog code into a seperate module.

closed 2023-03-16 by oharboe

wsnyder, 2023-03-16:
Ok, note that on my system it compiles in maybe 10 seconds, so chances are there's a gcc/clang performance bug that has been fixed.  gcc 11.3, clang 14.0.