verilator/verilator #4028: ibex_core_tracing is making verilator coredump

tenghtt, 2023-03-16:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
[ibex_comp_limit_blocks_fail.tar.gz](https://github.com/verilator/verilator/files/10986741/ibex_comp_limit_blocks_fail.tar.gz)
use 'make.sh' to reproduce

What 'verilator --version' are you using?  Did you try it with the git master version?
4.214/ 5.002/ 5.006/ 5.008

What OS and distribution are you using?
ubuntu 18.04/ win10

May we assist you in trying to fix this in Verilator yourself?
seems like an AST visit bug, with some treeNode freed too early, that's beyond my ability

wsnyder, 2023-03-16:
And note the step right before is V3DepthBlock, so yes, it makes sense that  -comp-limit-blocks would be what is causing it.

tenghtt, 2023-03-16:
yes captain, the smallest i can get is:
verilator_bin --debug -comp-limit-blocks 127 -Wno-MULTIDRIVEN -DRVFI ibex_pkg.sv ibex_tracer_pkg.sv ibex_tracer.sv --top-module ibex_tracer --trace-fst --cc

wsnyder, 2023-03-17:
Ok, first problem fixed which is -comp-limit-blocks can't split up a goto operation (from the "return" statements).
Looking at the CPP code it was ugly, so added a new optimization to reduce the number of labels.

closed 2023-03-17 by wsnyder