verilator/verilator #4029: Signal names with --report-unoptflat

T-Platz, 2023-03-16:
I'm currently debugging a circular combinational loop reported by Verilator as a `%Warning-UNOPTFLAT` with the `--report-unoptflat` flag. This generates a .dot file which I visualized using the `dot` command. However, this graph only contains a loop of eight nodes n0, ..., n7, all labeled with c1. Sadly, this doesn't really help, as I still don't know where exactly the loop is. Is there a way to include signal names and/or line numbers in the .dot file?

wsnyder, 2023-03-16:
When it built the subgraph it was missing cloning some information. Please try the new trunk.

wsnyder, 2023-03-16:
commit b2ced6ff1 - mentioned wrong bug number in message

closed 2023-03-16 by wsnyder