verilator/verilator #4032: SV: default clocking leads to syntax error later

svenka3, 2023-03-16:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
```verilog
module m;
  bit clk;

  default clocking dcb @(posedge clk);
  endclocking : dcb 

  function void build();
  endfunction : build

endmodule : m

```

What 'verilator --version' are you using?  Did you try it with the git master version?
> Verilator 5.009 devel rev v5.008-27-g1f1d9312d

What OS and distribution are you using?
> WSL

May we assist you in trying to fix this in Verilator yourself?
> Will attempt, but I am not much familiar with Verilator source code and YACC etc.

wsnyder, 2023-03-16:
This code works for me, please try it using the current master.

svenka3, 2023-03-16:
Thanks for qucik revert. I did a git pull, make, make install - I still see the issue at my end:

verilator -V
> Verilator 5.009 devel rev v5.008-52-g66e4656a8

wsnyder, 2023-03-16:
Oh, sorry, I see now it fails without --debug, but passes with it. That's very unusual I'll look into it.

closed 2023-03-17 by wsnyder