verilator/verilator #4033: always_comb unique else if with --assert will cause linter error

Mr-Bossman, 2023-03-17:
This is a similar issue to #3088

```systemverilog
module test (
	input logic in,
	output logic out
);
	always_comb begin
		unique if(in == 1) out = 1;
		`ifdef BUG
		else if (in == 0) out = 0;
		`endif
		else out = 1;
	end
endmodule

```
to test run:
```
$ verilator --lint-only --assert -DBUG test.v
%Warning-LATCH: test.v:5:2: Latch inferred for signal 'out' (not all control paths of combinational always assign a value)
                          : ... Suggest use of always_latch for intentional latches
    5 |  always_comb begin
      |  ^~~~~~~~~~~
                ... For warning description see https://verilator.org/warn/LATCH?v=5.006
                ... Use "/* verilator lint_off LATCH */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```
NOTE: this bug occurs only with --assert and an elseif 

Verilator 5.006 2023-01-22 rev (Debian 5.006-2)
also on 5.007

Mr-Bossman, 2023-03-18:
I have no experience with this code base, but I can try. Would the changes most likely be similar to the original fix?

wsnyder, 2023-03-18:
The other fix disabled the warning in certain places.  For this one, 
1. make a test case (see docs/internals.rst)
2. use --debug on the test to see what the .tree (AstNodes) look like
3. likely see that the "IF" AstIf statements have some pattern of if/elses that V3Active doesn't handle correctly and fix it.
4. "make test" to check all regression tests

Mr-Bossman, 2023-03-18:
Hmm so if we replace `always_comb` with `always_latch` it works but, in 

`Vtest_048_active.tree` we still get but nothing is printed to the console
```
%%Error: test.v:6: Assertion failed in %m: 'unique if' statement violated\n
```

wsnyder, 2023-03-18:
always_latch by definition suppresses LATCH.  Assertion is based on "unique" and unrelated to LATCH, but certainly could be a common bug.

Mr-Bossman, 2023-03-18:
when `m_assert` is true isBoundsCheck gets changed 
```bash
$ verilator --lint-only -DBUG ../test.v --assert 
../V3Assert.cpp visit 266
checkifp IF 0x55bd990241e0 {d11ak} checkifp->elsesp(): IF 0x55bd99027ec0 {d11ak} u1=0x1
../V3Active.cpp visit 328
nodep = IF 0x55bd99027ec0 {d11ak}
nodep->isBoundsCheck() = 0
../V3Active.cpp visit 328
nodep = IF 0x55bd990241e0 {d11ak}
nodep->isBoundsCheck() = 0
%Warning-LATCH: include/./../test.v:8:2: Latch inferred for signal 'out' (not all control paths of combinational always assign a value)
                                       : ... Suggest use of always_latch for intentional latches
    8 |  always_comb
      |  ^~~~~~~~~~~
                ... For warning description see https://verilator.org/warn/LATCH?v=5.009
                ... Use "/* verilator lint_off LATCH */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
$ verilator --lint-only -DBUG ../test.v
../V3Assert.cpp visit 266
checkifp IF 0x5645a56941a0 {d11ak} checkifp->elsesp(): IF 0x5645a5697e80 {d11ak} u1=0x1
../V3Active.cpp visit 328
nodep = IF 0x5645a5697e80 {d11ak}
nodep->isBoundsCheck() = 1
```

Mr-Bossman, 2023-03-19:
yes using gdb @ https://github.com/verilator/verilator/blob/master/src/V3Ast.cpp#L790 it will change the value to false. seems like a casting error, but if `m_isBoundsCheck` to the `AstNode` class it still changes the value.

Mr-Bossman, 2023-03-19:
Ok here is the fix I came up with. I'll send a PR tomorrow.
There is another bug I found with the `AstNodeIf` copy constructor that should be fixed.
 Also as seen in the prev comment the check for `isBoundsCheck` if done on  `checkifp->elsesp()` not  `checkifp`
```diff
diff --git a/src/V3Assert.cpp b/src/V3Assert.cpp
index 97259fc5f..84f447237 100644
--- a/src/V3Assert.cpp
+++ b/src/V3Assert.cpp
@@ -263,6 +263,9 @@ private:
                 = new AstIf{nodep->fileline(), new AstLogNot{nodep->fileline(), ohot},
                             newFireAssert(nodep, "'unique if' statement violated"), newifp};
             checkifp->isBoundsCheck(true);  // To avoid LATCH warning
+            AstNodeIf* const nextifp = dynamic_cast<AstNodeIf*>(checkifp->elsesp());
+           if (nextifp)
+                nextifp->isBoundsCheck(true); // To avoid LATCH warning
             checkifp->branchPred(VBranchPred::BP_UNLIKELY);
             nodep->replaceWith(checkifp);
             pushDeletep(nodep);
diff --git a/src/V3AstNodeOther.h b/src/V3AstNodeOther.h
index ed0bf5098..d78cbd708 100644
--- a/src/V3AstNodeOther.h
+++ b/src/V3AstNodeOther.h
@@ -415,6 +415,7 @@ protected:
         this->addElsesp(elsesp);
         isBoundsCheck(false);
     }
+    AstNodeIf(AstNodeIf& node) : AstNodeStmt::AstNodeStmt(node){}
 
 public:
     ASTGEN_MEMBERS_AstNodeIf;
```

Mr-Bossman, 2023-03-19:
> Hmm so if we replace `always_comb` with `always_latch` it works but, in
> 
> `Vtest_048_active.tree` we still get but nothing is printed to the console
> 
> ```
> %%Error: test.v:6: Assertion failed in %m: 'unique if' statement violated\n
> ```
> 
> the problem is this is not a latching block and if built without the unique it gives an error complaining about it being a non-latch

it also fixes this issue

Mr-Bossman, 2023-03-21:
This patch seems to fix it, should look for more instances like this
```diff
diff --git a/src/V3Const.cpp b/src/V3Const.cpp
index da4a74f62..abe512fef 100644
--- a/src/V3Const.cpp
+++ b/src/V3Const.cpp
@@ -2975,6 +2975,7 @@ private:
                 AstNode* const thensp = nodep->thensp()->unlinkFrBackWithNext();
                 AstNode* const elsesp = nodep->elsesp()->unlinkFrBackWithNext();
                 AstIf* const ifp = new AstIf{nodep->fileline(), condp, elsesp, thensp};
+                ifp->isBoundsCheck(nodep->isBoundsCheck()); // Copy bounds check info
                 ifp->branchPred(nodep->branchPred().invert());
                 nodep->replaceWith(ifp);
                 VL_DO_DANGLING(nodep->deleteTree(), nodep);
```

Mr-Bossman, 2023-03-21:
@wsnyder I made a pull to fix this https://github.com/verilator/verilator/pull/4054

wsnyder, 2023-03-21:
Thanks for fixing this and making a pull, it was not where I expected the bug to be!

closed 2023-03-21 by wsnyder