verilator/verilator #4036: SV: $info severity wrongly printed as "Assertion failed"

svenka3, 2023-03-17:
Thanks for taking the time to report this.
> $info is printed/reported as "Assertion failed" (this is misleading). 

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```verilog
`timescale 1ns/1ns
module m;
  initial begin
    #10;
    $info ("Info print");
  end
endmodule : m
```
LOG:
```log
[10] -Info: test.sv:5: Assertion failed in TOP.m: Info print
```
May we assist you in trying to fix this in Verilator yourself?

> Yes please, I believe this is easy one for a starter like myself :-)

svenka3, 2023-03-17:
I did some look into V3Assert.cpp and see that it always uses:

```cpp
 string assertDisplayMessage(AstNode* nodep, const string& prefix, const string& message) {
        return (string("[%0t] " + prefix + ": ") + nodep->fileline()->filebasename() + ":"
                + cvtToStr(nodep->fileline()->lineno()) + ": Assertion failed in %m"
                + ((message != "") ? ": " : "") + message + "\n");
    }

```
I believe one way to fix it would be to check the severity/displayType as:
```cpp
        if (nodep->displayType() == VDisplayType::DT_INFO) {
```

Am I on right track to a fix?

Thanks

wsnyder, 2023-03-17:
Yes, please make a pull request when you have it the way you want it, and note you will need to update the tests for the output, thanks.

svenka3, 2023-03-17:
With some hacking, I got the following:

```log
./obj_dir/Vaf_vlt_info
[10] -Info: af_vlt_info.sv:5: INFO in TOP.m: Info print
```
Changes done/discussion below:

I see the DisplayType getting replaced with DT_WRITE, why is this needed?
```cpp
V3Assert.cpp
 void replaceDisplay(AstDisplay* nodep, const string& prefix) {
        // AsFigo why nodep->displayType(VDisplayType::DT_WRITE);
```

I commented that for now. 

Step 2: I added additional arg to 

```cpp
 string assertDisplayMessage(AstNode* nodep, const string& prefix, const string& message, VDisplayType severity) {

        if (severity == VDisplayType::DT_INFO) {
```

If you believe this approach is OK, I will clean up and send a pull request.

Thanks

svenka3, 2023-03-17:
> Yes, please make a pull request when you have it the way you want it, and note you will need to update the tests for the output, thanks.

Great, which tests and what should I update please? Sorry to ask - am a newbie to this codebase/repo.

Thanks

wsnyder, 2023-03-17:
Run the tests and fix the ones that fail ;)   See the docs/internals.txt for more info on how that works, search for --gold.

Once V3Assert runs everything past there should be a normal display/write statement, so hence the type is changed.

I'd suggest "Info" not "INFO".

svenka3, 2023-03-17:
> Once V3Assert runs everything past there should be a normal display/write statement, so hence the type is changed.

So should I add that line at the end of this function?

```cpp
nodep->displayType(VDisplayType::DT_WRITE);

```

wsnyder, 2023-03-17:
Yes

svenka3, 2023-03-17:
Hope I don't bother you with such a tiny fix, but I see you already have the severity embedded, just had to knock-off Assertion Failed stuff:

```log
display print
[10] -Info: af_vlt_info.sv:6: in TOP.m: Info print

[10] %Warning: af_vlt_info.sv:7: in TOP.m: warning print

[10] %Error: af_vlt_info.sv:8: in TOP.m: Error print

%Error: af_vlt_info.sv:8: Verilog $stop
Aborting...
Abort
```
For code:
```verilog
`timescale 1ns/1ns
module m;
  initial begin
    #10;
    $display ("display print");
    $info ("Info print");
    $warning ("warning print");
    $error ("Error print");
    $fatal ("Fatal print");
  end
endmodule : m
```
Given a choice I would remove the "-" and "%" - but then this maybe a convention here, so will leave it as-is?

svenka3, 2023-03-17:
Note that SV LRM initially added these as "assertion display tasks" but later it allowed them to be used outside too (including elab-system-tasks, a separate topic), so removing "Assertion failed" is justified I believe.

svenka3, 2023-03-19:
One more related observation on this part of code in V3Assert - 

```cpp
else if (nodep->displayType() == VDisplayType::DT_ERROR
                   || nodep->displayType() == VDisplayType::DT_FATAL) {
            replaceDisplay(nodep, "%%Error");
```
I see that $error and $fatal are treated the same way - don't we want to distinguish and have separate counters for each severity and in case of $fatal - call $finish instead of $stop?

wsnyder, 2023-03-19:
The default Verilator handlers assume $stop aborts simulation as "bad", and $fatal as "good", therefore both fatal and error are the same $stop call.

svenka3, 2023-03-19:
I ran test_regress (cd test_regress, make) with and without my change on a CentOS machine. Below are my results:

```log
Without my change
TESTS DONE, FAILED: Passed 3080  Failed 14  Skipped 249  Time 644:24
```

With my change:
```log
TESTS DONE, FAILED: Passed 3062  Failed 32  Skipped 249  Time 644:43

```

Now, how do I triage this? Are there steps documented - if so please point me to the same. 

Thanks

svenka3, 2023-03-19:
> The default Verilator handlers assume $stop aborts simulation as "bad", and $fatal as "good", therefore both fatal and error are the same $stop call.

I propose we differentiate the message atleast - Error vs Fatal as prefix. Is that OK?

wsnyder, 2023-03-19:
See docs/internals.rst, generally you run each test with --gold to update each result.

Yes, I agree the text for error/fatal should differ.

wsnyder, 2023-03-24:
Thanks for the patch

closed 2023-03-24 by wsnyder