verilator/verilator #4038: SVA: Control vacuous success

svenka3, 2023-03-17:
What would you like added/supported?
> SVA pass action block gets triggered even on vacuous success. While that's per LRM, it is one of the most annoying things and most commercial simulators (if not all) do "$assertvacuousoff" internally today (Wasn't the case in the early days of SVA). It basically makes the pass-action blocks useless without such "feature". 

Consider:
```verilog

    a1 : assert property (
    @(posedge clk)
    a |=> !a)
    begin
     $info ("Pass");
    end 
    else
      $error("Failed");
```
In all cycles when "a" is 0, the SVA label a1 passes vacuously. This leads to following print:

```log
[5.000 ns] -Info: test.sv:12: in TOP.m.a1: Pass
[10.000 ns] -Info: test.sv:27: in TOP.m: a: 1
 15.000 ns a is 1
[15.000 ns] -Info: test.sv:12: in TOP.m.a1: Pass
 25.000 ns a is 1
[25.000 ns] -Info: test.sv:15: in TOP.m.a1: Error
[30.000 ns] -Info: test.sv:31: in TOP.m: a: 0
 test.sv:33: Verilog $finish
```

The above is from a controlled test to minimize the pass-action prints. In the above log, the first print Pass - is vacuous, second one is real success.

Now, as a typical user, I want the real success to be printed, the vacuous ones to be turned-off. SVA has $assertvacuousoff/on, but that's not yet supported in Verilator. Also, even if that works, user needs to do it - instead users expect this from the tool these days as the default behaviour. Is this not too complex to add here please?

May we assist you in trying to fix this in Verilator yourself?

> Yes, will attempt.

wsnyder, 2023-03-17:
I would suggest implementing all of $asserton/off, $assertpasson/off, $assertfailon/off, $assertnonvacuouson/off and these should just call a new implemented $assertcontrol.

I'm open to whatever subset of $assertcontrol features you need for your application.  I'd suggest starting relatively small (e.g. no scopes) for the first pull request..

svenka3, 2023-03-19:
> just call a new implemented $assertcontrol

By that you mean this assertcontrol is implemented inside Verilator? I couldn't locate that in src/* through a grep. Or you meant to refer to the LRM's addition?

Thanks

wsnyder, 2023-03-19:
No, I mean, don't implement just $assertvacuousoff/on, but rather implement $assertcontrol (or at least pieces of it).

ArturBieniek4, 2026-09-02:
This was already addressed by https://github.com/verilator/verilator/pull/7807. In case any shape you need is still not working, please open a separate narrowly scoped issue. 
@svenka3

closed 2026-09-02 by ArturBieniek4