verilator/verilator #4040: $monitor - ignore $time changes (LRM compliant)

svenka3, 2023-03-17:
Thanks for taking the time to report this.

What would you like added/supported?
> I see earlier issues/discussions that $monitor was not even supported. I see it now works and with some look in V3Assert I see Verilator replaces $mointor with always_comb. But this poses a problem as we get prints on every timestamp (sort of explainable given the always_comb). Just wondering if it is easy enough to fix this (Will need to filter out the sensitivity list and knock-off $time).

For sake of completeness, below is LRM ref:

> When a $monitor task is invoked with one or more arguments, the simulator sets up a mechanism whereby
each time a variable or an expression in the argument list changes value—with the exception of the $time,
$stime, or $realtime system functions—the entire argument list is displayed at the end of the time step as

Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)
```verilog
module m;

  bit clk, a, b;

  always #10 clk = ~clk;

  initial begin
    $monitor ("%t a: %b b: %b", $time, a, b);
    #100;
    a = 1;
    #10;
    b = 1;
    #10;
    a = 0;
    #10;
    b = 0;
    #100;
    #10;
    $finish (2);
  end
endmodule : m
```

May we assist you in trying to fix this in Verilator yourself?

> Yes please

wsnyder, 2023-03-17:
I'm a bit surprised people still use $monitor versus waves, but please make a pull request to fix this.

Presumably you will need to make variables and code to look for a change, that is "if (vara__Vpremonitor != vara || varb__Vpremonitor != varb) { .... display ....; vara = vara__Vpremonitor; varb = varb__Vpremonitor"

svenka3, 2023-03-19:
> I'm a bit surprised people still use $monitor versus waves, but please make a pull request to fix this.
>
> Presumably you will need to make variables and code to look for a change, that is "if [...]

Thanks for the code snippet suggestion. Where exactly does this fit in? I couldn't find any code using vara__ etc. so asking for more hints here.

wsnyder, 2023-03-19:
That was just example signal names.

wallento, 2023-05-02:
What @wsnyder described was the code generated. This would be generated in V3Assert, a good starting point is https://github.com/verilator/verilator/blob/master/src/V3Assert.cpp#L434
There you would generate the logic to track updates to the variables in your monitor statement, by tracking the old value and the new value and only monitor on change. So the given example code is the code that in the end roughly ends up in your generated C++. Generating the nodes in the Ast is a bit tricky, but it can be relatively straight forward I think. Please correct me if I am wrong, @wsnyder..

wsnyder, 2025-04-06:
Fixed last July with a2506a6f96f7a93d6efa715831007a49fce4341c

closed 2025-04-06 by wsnyder