verilator/verilator #4041: WIDTH warning from expression inside array access

leonardt, 2023-03-18:
Initial discussion in https://github.com/verilator/verilator/issues/3374

Verilog
```verilog
module Foo(
  input  [7:0] I,
  output [7:0] O);

  wire [1:0][7:0] _GEN = {{I - 8'h1}, {I + 8'h1}};
  assign O = _GEN[I != 8'hFF & I[0]];
endmodule
```

lint result
```
❯ verilator --lint-only Foo.v
%Warning-WIDTH: Foo.v:7:30: Operator AND expects 32 or 4 bits on the LHS, but LHS's NEQ generates 1 bits.
                          : ... In instance Foo
    7 |   assign O = _GEN[I != 8'hFF & I[0]];
      |                              ^
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
%Warning-WIDTH: Foo.v:7:30: Operator AND expects 32 or 4 bits on the RHS, but RHS's SEL generates 1 bits.
                          : ... In instance Foo
    7 |   assign O = _GEN[I != 8'hFF & I[0]];
      |                              ^
%Error: Exiting due to 2 warning(s)
```

Verilator 4.200 2021-03-12 rev v4.110-43-g5022e81af (will try updating to latest version but seems like it's reproducible by @wsnyder)

macOS Ventura 13.2.1 (22D68) 

Happy to help if you can point to the relevant code

leonardt, 2023-03-18:
Noting that simulation seems to behave as expected for the tested examples if `-Wno-width` is used

leonardt, 2023-03-18:
Just tested with `Verilator 5.006 2023-01-22` and seems to be in a similar state (warning is produced, simulation tests passing)

wsnyder, 2023-03-18:
Yes, attached is a test I made to double check it is making correct values. Rename to .v in test_regress/t, and cp test_regress/t/t_EXAMPLE.pl to test_regress/t/t_struct_packed_2d.pl

[t_struct_packed_2d.v.txt](https://github.com/verilator/verilator/files/11008390/t_struct_packed_2d.v.txt)

Given the results are correct, the problem is likely the computation of the expected width passed down to the operations is using the wrong array index. This would limit the problem to be in V3Width or V3WidthSel.  Use --debug to see the nodes that are causing the warning, then you can use a debugger or add prints at the warning to trace where the wrong sizes are coming from.