verilator/verilator #4042: Verilator built with MSVC crashes with unlock of unowned mutex

nuudlman, 2023-03-18:
Verilator 5.009 master built with CMake 3.25 and Visual Studio 2022, RelWithDebInfo.

Just running the executable is enough for it to crash on exit:
[image]

Call stack:
[image]

The bad unlock seems to happen when V3ThreadPool::resize() is called with size 0. Maybe the code to start the new jobs causes the destructor to fail?

wsnyder, 2023-03-18:
We have MSWIN regression actions on each commit which are passing, so it seems something related to your system/version/etc, can you debug further?

nuudlman, 2023-03-19:
I can try and look, but I'm not familiar with multithreaded programming. 

I'm running the latest version of windows 11 and building against the latest windows sdk/crt (10.26xxx).

nuudlman, 2023-03-24:
It seems to been the way I was building. When I built the same way as the ci script (clone winflexbison, cmake .. install prefix without spaces, cmake build release, set env var through system settings, clone verilator, etc) it seems to work. I have no idea why using visual studio to configure cmake causes it to fail in such a spectacular way.

nuudlman, 2023-03-24:
BTW the windows CI artifacts are broken, it always errors out with not being able to find verilator_std.sv, even though the file exists. Building from source fixes the issue.

closed 2023-03-24 by nuudlman