verilator/verilator #4043: Links to docs/internals are broken

timsnyder, 2023-03-18:
I've noticed that in the latest docs at https://verilator.org/guide/latest/contributing.html the links to `doc/internals` are pointing to a URL that returns 404 (https://verilator.org/guide/latest/internals.rst)

I see internals.rst in https://github.com/verilator/verilator/blob/master/docs/internals.rst.

1. Is something supposed to be publishing `internals.rst` to https://verilator.org/guide/latest/internals.rst and broken in the docs build?
2. If `internals.rst` is supposed to only live in the code tree, do you want the rendered docs to point to https://github.com/verilator/verilator/blob/master/docs/internals.rst?

I'd be happy to try and help with a PR that fixes these links.  Just let me know how you're intending for internals.rst to handled by the rendered documentation.

wsnyder, 2023-03-18:
I think internals just got forgotten in that build.  I'd be ok with just making a hard URL link as it seems to render ok in github.  Would you like to make a pull?

timsnyder, 2023-03-18:
I'm indifferent about making the PR.  If it is helpful for you, I can definitely submit one.  If you'd rather just fix it, my feelings won't be hurt. :)

In reading your comment, I'm not sure if you mean that you'd prefer it to be a part of the build?  It will definitely render better if you pass it through the full flow.  GH's preview is definitely "ok".

closed 2023-03-18 by wsnyder