verilator/verilator #4045: MULTIDRIVEN with task and default driver regression?

em2machine, 2023-03-18:
Thanks for taking the time to report this.

With Verilator version **Verilator 5.005 devel rev v5.004-86-g30fd49e21** the below testcase builds and simulates as expected (at least to me ;) ).  
```
/opt/verilator.prod/bin/verilator --binary --timing task_test.sv

echo "" > Vtask_test__ALL.verilator_deplist.tmp
Archive ar -rcs Vtask_test__ALL.a Vtask_test__ALL.o
/usr/bin/clang++  -Wl,-U,__Z15vl_time_stamp64v,-U,__Z13sc_time_stampv   verilated.o verilated_timing.o verilated_threads.o Vtask_test__ALL.a    -pthread -lpthread   -o Vtask_test
rm Vtask_test__ALL.verilator_deplist.tmp

XYZ@Mac-Studio bugs % ./obj_dir/Vtask_test
- task_test.sv:41: Verilog $finish
done

```

With Verilator version (latest master) **Verilator 5.009 devel rev v5.008-71-gf2aac8c49** I receive the following:

```
%Warning-MULTIDRIVEN: task_test.sv:13:7: Variable written to in always_comb also written by other process (IEEE 1800-2017 9.2.2.2): 'l0'
                                       : ... In instance m_tb.m
                      task_test.sv:13:7: 
   13 |       l0 = 'b0;
      |       ^~
                      task_test.sv:9:7: ... Location of other write
    9 |       l0 = 'b1;
      |       ^~
                      ... For warning description see https://verilator.org/warn/MULTIDRIVEN?v=5.009
                      ... Use "/* verilator lint_off MULTIDRIVEN */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```

Please refer to the testcase below.  My perspective is that the above error is not correct in this case.  Since the task is called inside the same process as the default driver, this shouldn't be multidriven?  I.e. "l0" is ultimately driven by both the task and the default driver inside the same process.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
```

module mod #()(
   input logic sel
   ,output logic val
);

   logic l0;

   task do_stuff();
      l0 = 'b1;
   endtask

   always_comb begin
      l0 = 'b0;

      if(sel) begin
         do_stuff();
      end
   end

   assign val = l0;

endmodule

module m_tb#()();

   logic sel;
   logic val;

   mod m(
      .sel(sel)
      ,.val(val)
   );

   initial begin
      sel = 'b0;
      #10;
      sel = 'b1;
      #10;
      sel = 'b0;
      #10;
      $finish;
   end

   final begin
      $display("done");
   end;

endmodule
```

What 'verilator --version' are you using?  Did you try it with the git master version?

Please see above

What OS and distribution are you using?

```
XYZ@Mac-Studio bugs % uname -a
Darwin Mac-Studio.local 22.2.0 Darwin Kernel Version 22.2.0: Fri Nov 11 02:03:51 PST 2022; root:xnu-8792.61.2~4/RELEASE_ARM64_T6000 arm64
```

May we assist you in trying to fix this in Verilator yourself?

wsnyder, 2023-03-18:
I'm not sure why you didn't get the warning on older versions, but Verilator does not yet analyze the flow from a process into a task/function to determine when warn or not.  If you'd like to improve V3Undriven.cpp to handle this it would be helpful.

phelter, 2024-03-04:
I'm seeing the same issue with migrating from verilator 5.005 to 5.022.

Additional false-positives for MULTIDRIVEN nets are in there when I have an interface task that drives a signal (only if called) and then another `always_comb` or `always_ff` block that drives the signal of an interface.

I think the best thing to do would be to remove all MULTIDRIVEN checks for tasks at this time to remove the false positives for now and then when support for diving into a task is supported in verilator.

Tasks should not be considered part of the MULTIDRIVEN check unless they are used.

closed 2026-01-01 by wsnyder