verilator/verilator #4046: Lock hang with debug

wsnyder, 2023-03-18:
@kamilrakoczy after #3805 was merged it seems with --debug most internal errors that hit a V3Broken error also, now hang, due to waiting for a lock.  For example, apply below and run test_regress/t/t_EXAMPLE.pl --debug.

[patch.txt](https://github.com/verilator/verilator/files/11009399/patch.txt)

wsnyder, 2023-03-18:
In addition to fixing, should we avoid all lock waits when single threaded?

kamilrakoczy, 2023-03-20:
Thanks for information. I looking at it.

kamilrakoczy, 2023-03-20:
It happens, when we are already printing error message, but due to some check, we are trying to print another error message. 
As we are storing all information about error message inside static variables/singleton, we shouldn't do such things, but it only happens after printing original message, so this should be fine.

I've fixed this deadlock with usage of recursive mutex, I'm also looking at option of disabling all lock waits when single threaded.

closed 2023-05-11 by wsnyder