verilator/verilator #4056: Capture code profiling for my source code

asif-ahd, 2023-03-22:
How can I capture code profiling of my source code in Verilator. The steps in the documentation are not clear and I am stuck. Thanks in advance. 
I am using version v5.007.

wsnyder, 2023-03-22:
Where are you stuck? e.g. what did you do so far and what error or otherwise are you getting?

asif-ahd, 2023-03-22:
So i am executing file source code in binary mode and I added the option _--prof-cfuncs__ while verilating. On completing the build a binary file _gmon.out_ in created in my current work area. Then what's next ? There are no clear examples to proceed and get the profile results. Thanks

wsnyder, 2023-03-22:
See in the distro test_regress/t/t_profcfunc.pl

We'd welcome a pull request to improve the docs as ypu figure this out

closed 2023-03-22 by wsnyder