verilator/verilator #4057: Verilated code uses variable before declaration

Sustrak, 2023-03-22:
Hi,

While using Verilator in a large project, the verilated code has a C++ error where a variable is used before declared. I can not provide with the SystemVerilog file to reproduce the error, so I'll try to describe it by posting the generated code, sorry for the inconvenience.

```c++
__Vilp = 0U; // Varable used without declaration
while ((__Vilp <= 0x2fU)) {
    // GENERATED CODE
}
/* GENERATED CODE */
......
......
......
/* GENERATED CODE */
CData/*0:0*/ __VicoContinue;
VlTriggerVec<13> __VpreTriggered;
IData/*31:0*/ __VnbaIterCount;
CData/*0:0*/ __VnbaContinue;
vlSelf->__VicoIterCount = 0U; 
__VicoContinue = 1U; 
IData/*31:0*/ __Vilp; // This declaration should move up
```
The verilator version I'm using is: `Verilator 5.008 2023-03-04 rev v5.008`
And the OS distro: 
```
LSB Version:	:core-4.1-amd64:core-4.1-noarch
Distributor ID:	Rocky
Description:	Rocky Linux release 8.7 (Green Obsidian)
Release:	8.7
Codename:	GreenObsidian
```

wsnyder, 2023-03-23:
findCreateVarTemp makes this signal, maybe try adding at the front of the cfunc, e.g. instead of `cfuncp->addInitsp(varp);`

something like 
`if (cfuncp->initsp()) cfuncp->initsp->addNextHere(varp) else cfuncp->addInitsp(varp);
`

If this works please make a pull request so we can fix it for good.

wsnyder, 2023-03-24:
Thanks for the pull.

closed 2023-03-24 by wsnyder