verilator/verilator #4059: Incorrect evaluation of combinational circuit with optimization -O3

StefanoBettelli, 2023-03-22:
I have a simple combinational circuit (generated by yosys), where one bit is evaluated incorrectly if verilator optimisation is turned on. I am attaching a SystemVerilog file and a minimal C++ testbench showing the problem.

```
>> /usr/local/bin/verilator --version
Verilator 5.009 devel rev v5.008-81-g277bd67f7

rm -rf obj_dir/ && /usr/local/bin/verilator -Wall -O3 --trace --trace-underscore --prefix DUT --top-module mymodule mymodule.sv --cc --exe --build testbench.cpp && ./obj_dir/DUT
Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

What 'verilator --version' are you using?  Did you try it with the git master version?

What OS and distribution are you using?

May we assist you in trying to fix this in Verilator yourself?

wsnyder, 2023-03-23:
-fno-const-bit-op-tree works around this, @yTakatsukasa perhaps you could look or if not suggest hints for @StefanoBettelli to debug?

StefanoBettelli, 2023-03-23:
After intensive exploration of the parametric space of the circuit (with different input sizes, strides and reduction operators [AND, OR or XOR]) it seems that the option `-fno-const-bit-op-tree` prevents incorrect code generation (whereas before I had several, but not all, parametric combinations with **XOR** showing a problem).

yTakatsukasa, 2023-03-25:
Thanks for the precise small test case.
I think I understand what's going on.
I will send a PR.

closed 2023-03-26 by wsnyder

StefanoBettelli, 2023-03-26:
Thanks for the quick fix! After extensive testing I can confirm that the last commit fixed all errors in my parametric circuit.

PR #4070 (2023-03-26): Fix incorrect optimization of bit op tree
Fixes #4059

The issue was introduced in #3459 when fixing #3445.

Push a test to see if the test properly reproduces the issue.
Then I will push the fix.

ext tests passes.

I found further optimization opportunity, but it is not included in this PR. I'd like to keep this PR just bugfix.