verilator/verilator #4065: Interface + generate begin does not compile

svenka3, 2023-03-24:
Thanks for taking the time to report this.

> When using "begin/end" in generate inside interface, I get a compile error. 

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```verilog
interface if_1;

  generate begin 
    wire my_id = 1;
    initial begin
      $display ("Hello: %0d", my_id);
    end
  end
  endgenerate

endinterface : if_1

module top;
  if_1 u_if_1();

  initial begin
    #100 $finish (2);
  end
endmodule : top
```

LOG:
```log
%Error: af_vlt_intf_gen.sv:3:12: syntax error, unexpected begin
    3 |   generate begin
      |            ^~~~~
%Error: af_vlt_intf_gen.sv:8:3: syntax error, unexpected end
    8 |   end
```
Notes:

1. The same code works fine if we replace interface with module
2. It also works with interface, if I change"generate begin" to "generate"

What 'verilator --version' are you using?  Did you try it with the git master version?
> Verilator 5.009 devel rev v5.008-60-g85fd88ace

What OS and distribution are you using?
> WSL

May we assist you in trying to fix this in Verilator yourself?
> Yes please - will attempt

wsnyder, 2023-03-24:
My reading of the IEEE 2017 grammar is that "generate begin" is not legal syntax. That is, begin for a generate can only be used under a for, if, or case statement.

Note this is regardless of being in an interface or not.

If you use "generate/endgenerate" without "generate begin.... end endgenerate" it works.

svenka3, 2023-03-24:
But that works in other tools. Also even Verilator accepts "generate begin" inside module, see below:

```verilog
module if_1;

  generate begin 
    wire my_id = 1;
    initial begin
      $display ("Hello: %0d", my_id);
    end
  end
  endgenerate

endmodule : if_1

module top;
  if_1 u_if_1();

  initial begin
    #100 $finish (2);
  end
endmodule : top
```

wsnyder, 2023-03-24:
While this still seems outside IEEE, I see no reason we can't make them consistent.

closed 2023-03-26 by wsnyder