verilator/verilator #4066: SVA checker..endchecker support

svenka3, 2023-03-24:
Thanks for taking the time to report this.

What would you like added/supported?
> checker..endchecker - LRM added these as containers for SVA. Though it has few additional features/benefits, a reasonable starting point can be as simple as treating them as module (ideally interface, but maybe Verilator has special code for it - to pass as port items etc.). For sure "procedural checker instantiation" is a far fetch and won't work with "module" approach. Also making it as module will prevent checker being bound to interfaces. Maybe there is better way?

Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)

May we assist you in trying to fix this in Verilator yourself?

> Yes please, will attempt.

wsnyder, 2023-03-24:
The verilog.y parser was recently updated to correctly (in theory) parse checker/endchecker, so the next step is to create AstNode's correspoinding to whatever constructs are new (and the checker itself) and make any appropriate self-checking tests (the existing test is just parsing to get UNSUPPORTED messages).  

Thanks for looking into improving this.

svenka3, 2023-03-24:
Thanks for the encouragement - so do you agree with replacing checker with module as a first-cut implementation? Not sure if it is easy, but would like to try it.

wsnyder, 2023-03-24:
I suspect make a new AstChecker that inherits from AstNodeModule, add some tests, update verilog.y, and you will get the first implementation you describe.

wsnyder, 2025-04-07:
Very basic functionality is now there for checkers, if there's additional features you'd like (which now give UNSUPPORTED) please open a new bug with appropriate example tests.

closed 2025-04-07 by wsnyder