verilator/verilator #4068: CMake verilate, build phase invokation of verilator

Risto97, 2023-03-25:
Hello,

I am using CMake as my build system for RTL simulations.
Very often I have a scenario where I have a custom command that would generate Verilog files.
I am using custom_commands rather than execute_process, as I prefer that my files are generated during build phase, so I can benefit from dependency management.

I am writing this issue to ask if anyone else has a better way of solving this problem, or if my solution is good enough I can make a pull request.

https://github.com/verilator/verilator/blob/master/verilator-config.cmake.in#L140
I am using function verilate linked above.
The problem with this is that function verilate expect that files in SOURCES argument exist at time of cmake configuration.

Then verilate proceeds to generate cmake file with the list of Cpp sources:
https://github.com/verilator/verilator/blob/master/verilator-config.cmake.in#L225
The list of cpp files is generated at configure time, and I assume that this list depends on the content of verilog files.
Which breaks my first ideas to solve this problem, that is:

## First solution
Generate dummy verilog files during configure phase, so verilator doesnt fail looking for files, but it also need to have proper ports and parameters so it can be instantiated.
I was doing this initially, but it would happen that if the files become drastically different, compilation would fail.

I demonstrate this on this small example:
https://github.com/Risto97/verilator_cmake_issue

A dummy file is provided during verilating:
https://github.com/Risto97/verilator_cmake_issue/blob/master/vsrcs/apb_subsystem_dummy.v
It only defines module that has the same ports, so the verilating doesnt fail

The real file that should generated is the next:
https://github.com/Risto97/verilator_cmake_issue/blob/master/vsrcs/apb_subsystem.v

To run the example do this: (tested on cmake 3.25, lower the version if older)
```
git clone https://github.com/Risto97/verilator_cmake_issue.git
cd verilator_cmake_issue
mkdir build
cd build
cmake ../
make 
```
The compilation will fail now:
```
/usr/bin/ld: CMakeFiles/vlt.dir/CMakeFiles/vlt.dir/Vtb.dir/Vtb___024root__DepSet_ha183790c__0__Slow.cpp.o: in function `Vtb___024root___eval_settle(Vtb___024root*)':
Vtb___024root__DepSet_ha183790c__0__Slow.cpp:(.text+0x9a): undefined reference to `Vtb___024root___eval_triggers__stl(Vtb___024root*)'
collect2: error: ld returned 1 exit status
```
As the Vtb.cmake is the one that is made for dummy file.

### Reconfigure project when file changes:
One solution is to rerun the configuration of the project when the file changes.
If you uncomment the line below the compilation will be successful. But I am not a big fan of this solution
https://github.com/Risto97/verilator_cmake_issue/blob/master/CMakeLists.txt#L46

## Use cmake external project
My next solution is the one that I am currently using, and that is to use verilate function in an cmake external project.
This way external project can depend on the targets that generate verilog files.
This way verilate external project is not configured until files are generated with their final content.

Basically this is the external project CMakeLists.txt
https://github.com/Risto97/VeriSC/blob/master/cmake/verilator/verilator_ext_prj.cmake
And this is how I include it in the main project:
https://github.com/Risto97/VeriSC/blob/master/cmake/verilator/verilator.cmake#L32
It is not ideal but it was the best solution I found.

wsnyder, 2023-04-02:
I'm open to this but the other CMake users need to comment, e.g. @johnjohnlin @codelec

codelec, 2023-04-03:
Thanks to ExternalProject, I don't see any need to move the verilate from configure to build phase
I posted an example usecase for a similar scenario here https://github.com/verilator/verilator/issues/4007#issuecomment-1479498981

Risto97, 2023-04-03:
Sure ExternalProject solves the problem, but it is a workaround I think. Because in the end still you are generating a new CMake project, and you have quite a bit of additional things to do to import your compiled library and link to your testbench.
Like here: https://github.com/Risto97/verilator_cmake_issue/blob/e7349296f29f52aae2bdd74387273e5bb7b72208/example_solution/verilate.cmake#L106

My thought is that I don't see any advantages of having verilator executed in configure phase instead of build phase, but the other way around there are some advantages, so I am questioning which should be the default behaviour that verilator provides out of the box? 
To name them: proper dependency management of build target verilate compared to configure phase, possibility to generate .V files.

To demonstrate what needs to be done, I uploaded this example, which I use for my use cases now, and try to mimic verilate() function.
For example I also add functionality that if --main is passed (verilog only testbench), probably there will not be executable target already existing, so the function can create one (in my example it can use the generated main.cpp instead of adding __null.cpp...)
https://github.com/Risto97/verilator_cmake_issue/blob/e7349296f29f52aae2bdd74387273e5bb7b72208/example_solution/verilate.cmake#L51

In the case that you decide to add build phase verilate function, the old function can still stay, because the new one can just use it in the ExternalProject CMakeLists.txt, but another function verilate_build_phase (or whatever makes sense) can be added to allow users to use more easily in bigger projects.

codelec, 2023-04-08:
ExternalProject is a CMake feature very well described here https://cmake.org/cmake/help/latest/module/ExternalProject.html and not a workaround
Having existing verilate function co-exist with the proposed verilate_build_phase is a nice solution

Risto97, 2023-04-09:
I will prepare a pull request in that case

Risto97, 2023-04-10:
@dlschmidt Yes, take a look at this function:
https://github.com/Risto97/verilator_cmake_issue/blob/e7349296f29f52aae2bdd74387273e5bb7b72208/example_solution/verilate.cmake#L1
And this is the external project:
https://github.com/Risto97/verilator_cmake_issue/blob/e7349296f29f52aae2bdd74387273e5bb7b72208/example_solution/verilator/CMakeLists.txt

Please keep in mind this is not a final solution, although it will work in most cases, I will prepare a pull request with the better implementation. Keep in mind the hard coded path for VERILATOR_HOME in the file

phelter, 2023-04-28:
I have an alternative approach that I'm willing to merge into verilator if it is something of interest. I have it working fully on my own environment and use it pretty much all the time instead of the CMake Verilator commands.

1. Make Verilog and SystemVerilog first-class citizens within CMake - in otherwords define the appropriate Language definitions for Verilog and SystemVerilog.  (Done in my local environment):
  * CMakeDetermineVerilogCompiler.cmake
  * CMakeTestVerilogCompiler.cmake
  * CMakeVerilogCompiler.cmake.in
  * CMakeVerilogInformation.cmake
2. Use CMake pre-defined target information for gathering all of the information for verilator (and other simulators) - i.e. Cross Compileable - eg:

```cmake
add_library( rs_cells_hdl INTERFACE )
add_library( rs::cells::hdl ALIAS rs_cells_hdl )

target_sources( rs_cells_hdl
  INTERFACE
    hdl/cell_buf.sv
    hdl/cell_dff.sv
    hdl/cell_inv.sv
    hdl/cell_mux.sv
    hdl/cell_nand.sv
    hdl/cell_nor.sv
    hdl/cell_xnor.sv
    hdl/cell_xor.sv
)

target_include_directories( rs_cells_hdl
  INTERFACE
    hdl
)
```

3. Separate out all of the verilator_base code into it's own library - this has the assumption that for any given cmake project ALL verilated code will be compiled the same way - i.e. with the same set of common compile definitions:
```cmake
target_compile_definitions( verilator_base
          PUBLIC
            VM_COVERAGE=$<BOOL:${VERILATOR_COVERAGE}> # TODO: use if ifdef with this in code - bad.
            VM_SC=$<BOOL:${VERILATOR_SYSTEMC}>
            VM_TRACE=$<OR:$<BOOL: ${VERILATOR_TRACE_VCD}>, $<BOOL:${VERILATOR_TRACE_FST}>>
            VM_TRACE_VCD=$<BOOL: ${VERILATOR_TRACE_VCD}>
            VM_TRACE_FST=$<BOOL: ${VERILATOR_TRACE_FST}>
        )
```
If you want a different set - then you reconfigure the entire project and compile all various verilated code with these parameters.  These then are used and linked to all verilated code to ensure they are the same parameters used by all consumers.
The benefits are - this gets compiled once and only once and then consumed for multiple different testbenches at different layers of a given project - i.e. `top level`, `macro A`, `macro B`, `macro C`, etc...

4.  Setup directives for parsing and consuming the verilator coverage if it is selected.  Also done in cmake as a separate command.  This allows you to amalgamate verilog code coverage from multiple layers of test benches all into one.  VERY VERY HELPFUL !!!!

5. `Verilate()` command in CMake now becomes:
```cmake
  # See above example of defining an HDL_TARGET.
  add_library( rs_cell_nand STATIC )
  RSVerilateLib( rs_cell_nand
    HDL_TARGET rs::cells::hdl
    TOP_MODULE cell_nand
    VERILATOR_ARGS
      ${VERILATOR_DEFAULT_ARGS}
  )
```
Benefits - can amalgamate similar functional blocks into an HDL_TARGET but define multiple VerilatedLib targets from that single HDL_TARGET with different names - really nice for combining common things like SRAM models, and asynchronous cdc modules all in a single HDL_TARGET.

6. Allow CMake to provide you a graphical visualization of your HDL dependency tree.  Just like graphviz for C based static libraries - you get this for HDL.
7. Have it also setup that `clean` command deletes all of the verilated generated code - nice to have to start from scratch.
8. So everything is done at Build and not Configure time - BUT, since the list of files generated by Verilator is unique to what is being compiled - you must have a two stage build for initial builds.  The first is to perform the verilate and the second is to re-query the list of verilated code generated - this is automated now and generates a new list every time.  If the list doesn't change then no need to recompile.
```bash
cd build
ninja GenerateVerilatedCode  # Done for the initial builds.
ninja
```
Also note if you are only changing the C/C++ code surrounding the verilated code - you don't need to do step 1 over again.  Only if the HDL changes.

Here's the description of the RSVerilateLib function:
```rst
RSVerilateLib
----------------

Creates a verilated target from verilog and c code.
Uses standard CMake definitions to define the list of files and directories

Based off of: https://github.com/verilator/verilator/blob/v5.006/verilator-config.cmake.in
And fixes following issues:
* Ability to run verilator at the build stage and not the generate stage of cmake
  * Remove need to consume a local cmake file in the generated directory
  * Add Custom target GenerateVerilatedCode to have verilated code be able to execute on its own
    two stage run of build cmake due to FILE GLOB being updated, but if file names don't change and the
    files are regenerated - this is reduced to 1.
* Ability to have file changes re-run the verilator function
  * TODO: Sub modules still may not cause this to happen - i.e. files from include
* Ability to clean out generated files with clean
* Removal of superfulous properties - all defined in compile_definitions of verilator base
* Missing checks for invalid configuration between Verilator::base and VERILATOR_ARGS
* General clean-up

RSVerilateLib(
    TARGET_DEST <target destination to add all verilated code>
    HDL_TARGET <target containing hdl(verilog) to verilate>
    [TOP_MODULE <top module>]
    [TRACE_STRUCTS]
    [PREFIX <prefix string>] - default is "V"
    [TRACE_THREADS <number of threads>]

    [VERILATOR_ARGS <args>]
    [OPT_SLOW <slow arg> [<arg> ...]]
    [OPT_FAST <fast arg> [<arg> ...]]
)
External Variables :
  VERILATOR_BIN - binary for executing verilate_bin exe.

  In FindVerilogTest which uses these for Verilator::base
    VERILATOR_COVERAGE - enable Coverage
    VERILATOR_SYSTEMC - enable SystemC
    VERILATOR_TRACE_VCD - enable for trace via VCD
    VERILATOR_TRACE_FST - enable for trace via FST

RSVerilateUsedBy( TARGET <target name verilator code is consumed in> 
)
This adds the necessary static analysis bypasses for anything consuming verilated c++ code.
```

dlschmidt, 2023-04-28:
I highly appreciate this change and would be grateful if this could be merged into main.

Especially the following part would be very useful for incremental builds for testbenches:

> Also note if you are only changing the C/C++ code surrounding the verilated code - you don't need to do step 1 over again. Only if the HDL changes. One last thing - this is all additive so can allow the older version to stick around and possibly be deprecated at a much later date and then have this as an alternative approach to organizing your Verilog projects.

Risto97, 2023-04-28:
@phelter 
Really interesting approach. 
 I have a few questions though.

I am using INTERFACE libraries in a similar way.
The verilate function I have is operating on libraries instead of filesets.
To compose a hierarchical (dependent) model of IPs I use target_link_libraries to link them into a tree, and then on consumption, I have a custom tree traversal function to flatten the tree and extract verilog files from each library in the correct order.
Would this help with automatically inheriting sources and properties to linking libraries?

1. I assume this will allow you to set the TOOLCHAIN_FILE before the project call to set up the verilator as Verilog compiler?
One problem I noticed with cross-compiling with CMake in general is the notorious infinite loop when you want to change the 
compiler in the middle of project, cmake does not allow that, it makes you use an external project for cross-compiling. Maybe its only a problem if you are changing the compiler for the same language e.g. GCC x86 for GCC Risc-V.
I typically use different "Verilog compilers" within the CMake project, not only Verilator, maybe it would be a good idea to not use Verilator as the Verilog compiler, so not call it CMakeDetermineVerilogCompiler.cmake?
8. Hmm this 2 step process is a bit annoying, can GenerateVerilatedCode be a custom command that tracks verilog file changes and only run when needed automatically as a dependency to next target?

I am interested in this approach and wonder if it would solve some of the problems the external project approach has.
Is it possible for you to publish it on your fork so we can take a look how it works? Also with an example of usage if possible.

Thanks.

wsnyder, 2023-04-28:
>Separate out all of the verilator_base code into it's own library - this has the assumption that for any given cmake project ALL verilated code will be compiled the same way - i.e. with the same set of common compile definitions:

Note this should probably be verilated_base - as it's the runtime library.

It's worth noting that "for any given cmake project" is a correct and important part of the requirements.  As to why that's needed versus always making an e.g. libverilated.a across all projects, see #3507.

phelter, 2023-04-28:
@wsnyder -
>Note this should probably be verilated_base - as it's the runtime library.

Yes it's similar to verilator base.  But I'd suggest leaving that as code at the moment.  you have the variations defined in #3507: Reasons:
* You want to allow the user to define whether they are using their stdlib variant - (Clang or GNU).  No ABI issues if they pop up.
* You want to allow the user to build how they want to - I have typically used Release and Debug for verilog test, but also have used RelWithDebInfo, ASAN, TSAN, UBSAN, Coverage scenarios. - and integrating a lib that is not compiled with those has some dire consequences on debugging issues under those build parameters.

@Risto97 
>  Would this help with automatically inheriting sources and properties to linking libraries?

I do no ordering - just linking of `INTERFACE` libraries as you mentioned and CMake internal Properties of libraries to collect the file sets and include directories - Yes it's the way to go for combining multiple targets together in Cmake.  
The `INTERFACE` option would not be necessary if verilator could verilate a standalone `*.v` or `*.sv` and convert it into it's own cpp file and not perform the flattening that it does.  One could do that with defining the Parameters as Templates, but that would mean moving to a scenario where the basic structures such as input/outputs are no longer `uint32_t`'s or smaller, but a template-able size as well - not in the cards see: #4108.  So this is an easy way to get around that just defining the HDL targets as INTERFACE.

> One problem I noticed with cross-compiling with CMake in general is the notorious infinite loop when you want to change the compiler in the middle of project, cmake does not allow that, it makes you [...]
I typically use different "Verilog compilers" within the CMake project, not only Verilator, maybe it would be a good idea to not use Verilator as the Verilog compiler, so not call it CMakeDetermineVerilogCompiler.cmake?

I change the compiler between Clang and GNU all the time.  No issues there - this is a project wide thing.  I hardly ever use ExternalProject except when I need to have multiple different compilers in the same project - i.e. compile a test program for a verilated processor.  So not sure what you are referring to.
In terms of different "Verilog compilers" - Yes - I've got it setup to add the others Siemens Questa - (Formerly known as Mentor), Cadence NCSim, Synopsys , and opensource Icarus, slang - but haven't put in the time to integrate those quite yet since I haven't had a need. - to change the Verilog compiler on your initial Cmake configure step you add:

```bash
cmake -DCMAKE_Verilog_COMPILER=slang -G "Ninja" . - That's it.  Everything else is in the details of the compiler definition files.
```
I currently have `slang` and `verilator` there as I wanted to use slang as a means of linting.  The optimal thing would be to get CMake to own this portion of the definition, and have support from the various companies to define their own Compiler definitions, but they probably see this as a downside since their tool would be too easily swap-able for another one (no stickiness to using their tool).

>Hmm this 2 step process is a bit annoying, can GenerateVerilatedCode be a custom command that tracks verilog file changes and only run when needed automatically as a dependency to next target?

This approach already tracks the verilog file changes and only runs the `verilate_bin` on the appropriate `RSVerilateLib()` hdl library targets that have changes (or their dependencies have changes) - in other words does Incremental compiles and builds correctly.  Yes the target dependencies work similar to normal C/C++ libraries.  The reason you need  two stage process is because there isn't a 1:1 mapping of file names for each verilated solution.  If the [files generated](https://verilator.org/guide/latest/files.html) from a verilated set of verilog files were consistent -  i.e. removal of all `{.*}` and only allow a single `{prefix}` then it would be possible to do a one stage build.  But unfortunately that is not the case.

I only have the two stage the very first time after a `clean` or the Configure stage to ensure the fileset is consistent and defined.  Because the fileset feeds into the list of files needed for the Verilated C/C++ library target this is unfortunately necessary.  Yes a bit annoying but way less annoying than going back to the Configure stage every single time - or dealing with ExternalProjects which do *NOT* have the ability to incremental build (as mentioned above).

Yes I can get an example of it going if you like on a fork.  Unfortunately I'm super busy right now so it may take a while.

wsnyder, 2023-04-28:
>>Note this should probably be verilated_base - as it's the runtime library.
>Yes it's similar to verilator base. But I'd suggest leaving that as code at the moment. you have the variations defined in https://github.com/verilator/verilator/pull/3507: Reasons:

My point was only that it should be renamed from verilator_base to verilated_base.

Risto97, 2023-04-29:
@phelter 

> I change the compiler between Clang and GNU all the time. No issues there - this is a project wide thing. I hardly ever use ExternalProject except when I need to have multiple different compilers in [...]

Well exactly that, if you try to change CMAKE_LANG_COMPILER within the project it is not allowed because the value is cached, and you get this message, together with the loop:
```
You have changed variables that require your cache to be deleted.
Configure will be re-run and you may have to reset some variables.
The following variables have changed:
CMAKE_CXX_COMPILER= /usr/bin/c++
```
I kind of can get around it by setting the CMAKE_CXX_COMPILER explicitely before the testbench (host gcc) project call, and then later it kind off works when I change the toolchain for the cross compilation to another one. Although its probably not completely safe to do, so I am extra careful with comp and link flags.

But it appears that you maybe do not think of changing the CMAKE_VERILOG_COMPILER in the middle of project, but per configuration?
Meaning you would configure the project to run Verilator, then if you want to run Slang you reconfigure it again?
My workflow with different "verilog" compilers is that I have them all present as make targets, so I can run
`make some_tb_vlt` or `make some_tb_vcs`, `make soc_vivado` ...
It simplifies things a bit for when I need to explain to someone how to run something in the project, although maybe not completely consistent with CMake philosophy of compilers.

> Yes I can get an example of it going if you like on a fork. Unfortunately I'm super busy right now so it may take a while.

No worries whenever you have time.

Timmmm, 2023-05-25:
This would also fix #4007. However the downside of doing it in the build phase is that you don't have a C++ target in CMake at configure time, so you can't e.g. set compilation flags and so on in the same way that you would for a normal C++ library.

You can't do that because native CMake libraries need to know their sources at configure time, and you don't know the source files for the library until Verilator runs.

phelter, 2023-05-28:
@Risto97 

> Well exactly that, if you try to change CMAKE_LANG_COMPILER within the project it is not allowed because the value is cached, and you get this message, together with the loop:

Yes correct - for any given `CMake Configure` - you can only select one `CMAKE_<LANG>_COMPILER` for each <LANG> - if you want more then you create a separate build tree for each of the various configurations - swapping out different compilers. - each can be in their own separate `build` directory:

eg:
``` bash
# Build with clang and verilator
cmake -G "Ninja Multi-Config" -Bci_build/clang_verilator -DCMAKE_C_COMPILER=clang-14 -DCMAKE_CXX_COMPILER=clang++-14  -DCMAKE_Verilog_COMPILER=verilator -DCMAKE_-DFETCHCONTENT_BASE_DIR=build/deps ${CMAKE_BUILD_ARGS} .
# Build with gnu and slang
cmake -G "Ninja Multi-Config" -Bci_build/gnu -DCMAKE_C_COMPILER=gcc -DCMAKE_CXX_COMPILER=g++ -DCMAKE_Verilog_COMPILER=slang -DFETCHCONTENT_BASE_DIR=build/deps ${CMAKE_BUILD_ARGS} .
```

Note the Multi-Config allows you to build `Release, Build, RelWithDebInfo, ASAN, TSAN, UBSAN, Coverage` build types all in the same directory - but you must still have a single compiler definition - can't swap per build directory.

And then if you want to run something with gnu you either:
```bash
cd build/<compiler variant>
ninja
```
or via cmake execute - regardless of which build generator you use:
```bash
cmake --build build/<compiler variant> --config Debug
```

> My workflow with different "verilog" compilers is that I have them all present as make targets, so I can run `make some_tb_vlt` or `make some_tb_vcs`, make `soc_vivado` ...

Yeah - that's one way, but I tend to have one target eg `some_tb` and then the selection of how that gets compiled for various types of compilers is defined in the CMakeLists.txt eg:

```cmake

add_library( some_tb )

target_sources( some_tb
     $<$<COMPILE_LANG_AND_ID:Verilog,Verilator>:verilator_specific_file.h>
    some_tb_common_file.v
)

target_compile_options( some_tb
     $<$<COMPILE_LANG_AND_ID:Verilog,Slang>:"Some slang specific compiler options">
)
# Hope this explains the point.
```

And then no matter which way you compile it, you can always run some_tb or consume some_tb into another library and use it that way.  I believe this is more consistent with the CMake philosophy.

@Timmmm

>This would also fix https://github.com/verilator/verilator/issues/4007. However the downside of doing it in the build phase is that you don't have a C++ target in CMake at configure time, so you can't e.g. set compilation flags and so on in the same way that you would for a normal C++ library.

> You can't do that because native CMake libraries need to know their sources at configure time, and you don't know the source files for the library until Verilator runs.

You can with newer versions of cmake.  I've been doing this for some time now.

```cmake
  # VDIR = directory verilating into.
  # VERILATE_PREFIX = verilated code prefix for identifying the verilated file set.
  file(GLOB GENERATED_C_SOURCES LIST_DIRECTORIES false
       CONFIGURE_DEPENDS
       ${VDIR}/${VERILATE_PREFIX}_*.cpp
       ${VDIR}/${VERILATE_PREFIX}_*.h)

  target_sources(${TARGET_DEST}
    PRIVATE
      ${VDIR}/${VERILATE_PREFIX}.h # This links the generation of the add_custom_command above.
      ${VDIR}/${VERILATE_PREFIX}.cpp
      ${GENERATED_C_SOURCES})
```
Look up `CONFIGURE_DEPENDS` - which allows this specific function.  It may require multiple runs of build to work, but I separate them into a two stage build - first to verilate the code via a single target `GenerateVerilatedCode` and then second to build.  The glob always works.

What I'd prefer would be that the file list was known based on the verilator settings, but that's not quite the case since the number of files and their contents may vary - especially if using hier options and other things.

phelter, 2023-05-29:
@wsnyder and @Risto97 - started moving this from my local cmake to cmake and this repo:

Step 1- [CMake support for verilog](https://gitlab.kitware.com/cmake/cmake/-/merge_requests/8517) is now submitted as a PR.

So if you want to try this out yourself you can copy the added files into a `local_cmake/Modules` directory somewhere for now and then add in your toplevel CMakeLists.txt:

```cmake
list( APPEND CMAKE_MODULE_PATH "<root>/local_cmake/Modules")

# Before your project( xxxx ) call.
```

This will at least identify verilog as a known language.

I'll add the verilatelib now to a fork of this repo for you to try out.

I typically combine things in one of two ways (a level of hierarchy with a standalone testable unit - with one toplevel, or a level of hierarchy that collects common group of similar modules together) eg: directory/source hierarchy:

```txt
# Standalone block

block \
  hdl \
      block.sv
      block_subblock.sv
      block_subblock2.sv
  CMakeLists.txt
  test \
    block_testbench.sv  
       #(sometimes not needed - only when there are systemverilog interfaces - since these aren't supported at top)
    test_block.cpp
    CMakeLists.txt

# OR collection of similar modules

cells \
   hdl \
     cell_or.sv
     cell_and.sv
     ...
   CMakeLists.txt

# Common synchronizers for cross-clock-domain synchronization

sync \
   hdl \
     sync_signal.sv
     sync_rstb.sv
     sync_bus.sv
     sync_bus_handshake.sv
    ...
   CMakeLists.txt
```

For the `block/test/test_block.cpp` for me I'm most familiar with the C++ Test Framwork `GoogleTest` so use that with my homegrown UVM like drivers.  These drive everything via the Verilated C interface.  - Want to use the VPI interface but not quite there yet - reason is to allow same TestSuite to run on verilator and other simulators since verilator doesn't support gatelevel sims.

Example for the `cells/CMakeLists.txt then looks like:
```cmake
add_library( rs_cells_hdl INTERFACE )
add_library( rs::cells::hdl ALIAS rs_cells_hdl )

target_sources( rs_cells_hdl
  INTERFACE
    hdl/cell_buf.sv
    hdl/cell_dff.sv
    hdl/cell_inv.sv
    hdl/cell_mux.sv
    hdl/cell_nand.sv
    hdl/cell_nor.sv
    hdl/cell_xnor.sv
    hdl/cell_xor.sv
)

target_include_directories( rs_cells_hdl
  INTERFACE
    hdl
)

#--------------------------------------------------------------------
add_subdirectory(test)
```

And the `cells/test/CMakeLists.txt then looks like:

```cmake
set( _module_list buf dff inv mux nand nor xnor xor )
foreach( _mod IN LISTS _module_list )
  add_library( rs_cell_${_mod} STATIC )
  # Note need absolute path for coverage
  RSVerilateLib( rs_cell_${_mod}
    HDL_TARGET rs::cells::hdl
    TOP_MODULE
      cell_${_mod}
    VERILATOR_ARGS
      ${VERILATOR_DEFAULT_ARGS}
  )
  list( APPEND _module_libs "rs_cell_${_mod}" )
endforeach()
unset( _module_list )

add_library( rs_cells INTERFACE )
add_library( rs::cells ALIAS rs_cells )
target_link_libraries( rs_cells
  INTERFACE
    ${_module_libs}
)
unset( _module_libs )
````
This is primarily because there is no intermediate per-verilog module compile stage - you must verilate everything that you want as a top module in their own independent RSVerilateLib() target.  Which is fine, but would be much easier if the verilate step could be done on each verilog file independently and then the C files combined but I know that's not how verilator works.

Risto97, 2023-05-29:
@phelter thank you for sharing, great work.
I am looking forward to try this out.

I only have one problem with this approach and would like to hear your opinion.
I use CMake to make a flow for my hardware projects, let me give one example.

I might for example do yosys synthesis or optimization before running a simulator. Then I would have a dependency of running yosys before verilator, so in my case it was looking like this:

```cmake
add_library(some_lib INTERFACE
   <some_files...>
) 
if(SYNTH)
    yosys(some_lib )
endif()
```
Another CMakeLists.txt for testbench
```cmake

add_subdirectory("<path-to-rtllib>")
verilate(some_lib)
```
Or I might run sv2v converter, because some construct is not currently supported in verilator.
Functions yosys or sv2v will then replace the file list  in the some_lib with the generated files, so verilator will verilate those instead.
It is important in this case that I can change the Verilog compiler mid project, so I can have them run one after the other.
With C++ cross compilation, I found out a hacky way around the infinite loop, that if you set explicitly `set(CMAKE_CXX_COMPILER /usr/bin/g++)` before the first project call, the value doesn't get cached? and then I am allowed to change it again later in the project.
Maybe this will allow to change the compiler mid project? (Really hacky solution, and I don't like it though).

In your example, I can see how verilator, vcs, xcelium probably don't need to run in the same project. 
However I don't agree with slang, as in an environment I might decide to do linting with slang before running simulation.
This might be needed in a collaborative environment where people might forget to lint.
In this case I think that slang would be more useful as a function wrapped around add_custom_target/command.

Where do you draw the line to what is implemented as a CMake language compiler?

On unrelated note:

>  These drive everything via the Verilated C interface. - Want to use the VPI interface but not quite there yet - reason is to allow same TestSuite to run on verilator and other simulators since verilator doesn't support gatelevel sims.

I had similar idea, and I created some wrapper classes around different simulator features. The idea was to use the same SystemC-UVM testbench and be able to run different simulators on the same testbench. At the moment I had only verilator and vcs supported with limited features.
I plan to have this open-sourced, if you have the same idea and plan to open source this work, maybe we can combine efforts on this?

phelter, 2023-05-29:
Thanks @Risto97 - I mentioned this above this thread [here](https://github.com/verilator/verilator/issues/4068#issuecomment-1565888749) but I will typically create multiple build directories:

```cmake
# Build with clang and verilator
cmake -G "Ninja Multi-Config" -Bci_build/clang_verilator -DCMAKE_C_COMPILER=clang-14 -DCMAKE_CXX_COMPILER=clang++-14  -DCMAKE_Verilog_COMPILER=verilator -DCMAKE_-DFETCHCONTENT_BASE_DIR=build/deps ${CMAKE_BUILD_ARGS} .
# Build with gnu and slang
cmake -G "Ninja Multi-Config" -Bci_build/gnu -DCMAKE_C_COMPILER=gcc -DCMAKE_CXX_COMPILER=g++ -DCMAKE_Verilog_COMPILER=slang -DFETCHCONTENT_BASE_DIR=build/deps ${CMAKE_BUILD_ARGS} .
# Build for other functions
cmake -G Ninja -Bci_build/yosys -DSYNTH=ON -DFETCHCONTENT_BASE_DIR=build/deps ${CMAKE_BUILD_ARGS} .
```
Please review that comment above - it illustrates the `CMake philosopy` and how I think the full flow for VLSI development can be made to work within it is to reduce it into :
| Input  | steps (multiple) | Outputs |
|-------| -----------------|----------|
| hdl    |  Verification        | Coverage Results|
| hdl    |  Synthesis          | post synth structural verilog (psynth-net) |
| psynth-net | Layout | GDS2 / post layout structural verilog (playout-net) |
| playout-net | STA | Timing results |
| etc | etc | etc |

The Inputs should be static `input target` definitions with minimal change per cmake run.
The steps are selected via options - i.e. `RUN_SYNTHESIS`, or `BUILD_TEST`
And the above steps - I typically run independently via a CI yaml build script and don't attempt to run those all back-to-back using CMake, but you can do that if needed.

If you do want CMake to run them independently and in a certain order you can put all of the input hdl in a separate folder - eg `foo` and create those steps etc inside that directory's CMakeLists.txt and then the top-level is executing the above separate CMake commands in the order you require - or via `ExternalProject()` function - but you do need to have some layer above that organizes the inputs/outputs and steps mentioned in the table above.  Don't try and do it all in the project's cmake file though it will get very unruly very fast.  To allow switching in IDE's such as VSCode - I typically make a [cmake-variants](https://vector-of-bool.github.io/docs/vscode-cmake-tools/variants.html) to build those toplevel steps for Development purposes - also provides a great way to document your flow (outside of the CI/CD yaml files).

For the `sv2v` - steps if those are simple create this output file - and you know the filenames then those are easily added in the same cmake step via `add_custom_target` and define the output dependencies - they will be executed every time something changes and the outputs are in your build directory with a known location and filename to generate again and again.

>  However I don't agree with slang, as in an environment I might decide to do linting with slang before running simulation.

Yeah - for slang could treat it like some of the Some `C/C++` based tools like `clang-tidy/cpplint/iwyu/...` these essentially run in parallel with the compile and provide additional information because they require the compile -D defines etc to do that.  Might be able to do something similar to that, but didn't want to spend the effort at this point, and it is really the only open-source tool that compiles more like the commercial tools such as vcs/ncverilog/modelsim - so thought it might be better just to run it in that fashion.

> I had a similar idea.

Yes - I'd be interested.  I've done this before to the point of using it to drive lab equipment and it works really well - same test environment for both simulation and post-layout/tapeout validation.  (not quite as useful for FPGA's) but exteremely helpful - and basically the goal of what the industry calls `portable-stimulus`.

Risto97, 2023-05-31:
@phelter I sent you an email, so we keep this thread on topic

nickelpro, 2023-05-31:
Hey just wanted to say we're following this closely at the Purdue SOCET team and NYU Processor Design Team, we have a somewhat [jank set of cmake functions](https://github.com/NYU-Processor-Design/nyu-cmake/blob/main/cmake/nyu-verilog.cmake) to implement similar behavior right now.

We came to a similar conclusion as @phelter that the right answer was upstream CMake support but didn't have time/expertise to implement. Great work getting a PR up!

I haven't had time to go over everything just yet, but wanted to bring up a common use case we have:

We package `.sv` and related files similar to the way header FILE_SETS work, we `install(FILES)` each verilog language file that's been added to the target, fixup the paths, and associate them with the EXPORT target. This lets us manage dependencies with the typical CMake tools (FetchContent/vcpkg/find_package).

This isn't exactly the original issue, but it would be very nice if any native cmake support for SV supported the typical `install(TARGETS)` -> `install(EXPORT)` workflow for modern cmake packaging. I'll try to take a look at phelter's patch sometime this week, if people don't have objections to the idea.

Timmmm, 2023-06-01:
> It may require multiple runs of build to work,

@phelter Yeah that is not really going to fly I think. Also I misread this proposal - I don't think it fixes the issue of generated code in my case since the `.sv` generator I am using generates an unpredictable set of files (similar to how Verilator generates C++ files) and this proposal requires you to know the names of the `.sv` files in advance (though there is a `.f` file listing them all).

Risto97, 2023-06-09:
@Timmmm not ideal solution, but if you put verilator in external project, you can glob during configure phase for generated verilog files.

wsnyder, 2026-10-18:
Closing due to age, some good ideas here but without a PR champion not planned.

closed 2026-10-18 by wsnyder