verilator/verilator #4069: Benchmarks for Verilator?

burntfalafel, 2023-03-25:
Hi! Are there any reliable benchmarks I could use in Verilator for a research paper? 

I have tried [VTR datasets](https://docs.verilogtorouting.org/en/stable/vtr/benchmarks/) and [EPFL ones](https://github.com/lsils/benchmarks) but for both the verilog is in a netlist format / a format not supported by Verilator readily.

wsnyder, 2023-03-25:
See the Swerv cores that are part of https://github.com/verilator/verilator_ext_tests

closed 2023-03-25 by wsnyder

gezalore, 2023-03-25:
You can also consider lowrisc/OpenTitan, or OpenXiangShan/XiangShan which are the two I often use

wsnyder, 2023-03-25:
As bonus credit, consider submitting a pull to add OpenTitan and/or XiangShan to the verilator_ext_tests.

wsnyder, 2023-03-25:
Also note the test_driver script has a --benchmark option to print the times used by each step.

burntfalafel, 2023-03-26:
> As bonus credit, consider submitting a pull to add OpenTitan and/or XiangShan to the verilator_ext_tests.

I will have a look at them, thank you! But please go ahead with the PR for this as I cannot take the credit for your contribution :)