verilator/verilator #4077: Static error over sensitive

wsnyder, 2023-04-01:
@kbieganski #4072 seems to be mis-firing on code that was previously correct.  For example Ibex in the sv-tests.  Also a new early error in UVM that worked before

        virtual function uvm_root get_root();
                return uvm_root::m_uvm_get_root();
        endfunction

wsnyder, 2023-04-01:
Also ariane.  I'll commit to comment this error out for now, to avoid breaking those projects.

kbieganski, 2023-04-02:
Sorry about that :disappointed:. I'll try to do a proper fix.

RRozak, 2023-05-24:
The problem (at least in UVM) occurs because static methods defined outside a class (but predeclared in the class body) aren't properly marked as static. I started working on it and I hope that the problem stops occurring.

RRozak, 2023-05-25:
I found another example of false error, but it isn't related to extern declarations:
```
package my_pkg;
  function automatic int get_1;
     return 1;
  endfunction
endpackage

module t;
   initial $display("%d", my_pkg::get_1());
endmodule
```

wsnyder, 2025-04-07:
Test case passes on current master.

closed 2025-04-07 by wsnyder