verilator/verilator #4083: Select from non-array BASICDTYPE 'bit'

davidel, 2023-04-04:
OS:
```
MacOS Ventura 13.3
```

Verilator Version:
```
$ verilator --version
Verilator 5.008 2023-03-04 rev UNKNOWN.REV
```

Code:
```Verilog
module Test(CLK, XOUT);
  input logic CLK;
  inout logic [15: 0] XOUT[8][8];
  always @(posedge CLK)
  begin
    XOUT[1][3] = XOUT[0][5] + 1;
  end
endmodule // Test
```

Error:
```Shell
$ verilator -sv --lint-only -Mdir /tmp /tmp/test.sv
%Error: /tmp/test.sv:6:9: Select from non-array BASICDTYPE 'bit'
                      : ... In instance Test
    6 |     XOUT[1][3] = XOUT[0][5] + 1;
      |         ^
%Error: Exiting due to 1 error(s)
```

Even a simple assignment of a constant generates the same error (the RHS does not influence the error).
I have tested the above with *Vivado* and *Quartus*, and they seem OK with that syntax.

wsnyder, 2023-04-04:
Probably something related to the mix of packed and unpacked arrays, the type of the unpacked extraction seems to be seen as a unpacked number instead of the 2D packed array.

Perhaps you'd be willing to dig into V3Width/V3Width to see why this is?

wsnyder, 2023-04-05:
Oh, that's much later than I thought.  Something in V3Unknown needs improving to avoid this, the input tree going into V3Unknown seems ok, maybe V3Unknown needs some improvement for 2D unpacked arrays.

wsnyder, 2025-09-25:
This passes on master, was fixed sometime earlier.