verilator/verilator #4085: Perl perf issue in =~ (binding operator) on verilated.cpp

svenka3, 2023-04-04:
An off-topic (not Verilator issue) - on regression runs in test_regress.

On a CentOS I see t_dist_cppstyle.pl hangs for hours together. With some prints I located the issue to be around "binding operator" in Perl - for some big (not really big) file such as verilated.cpp

```perl
  printf ( "$filename $message $pattern %0d \n", length($buffer));
    while ($buffer =~ s/.*?^($pattern)//sm) {
      print "AsFigo2: $filename $message \n";
```
I got below:
```log
 ../include/verilated.cpp 'virtual' keyword is redundant on 'override' method (?^:[^/]*virtual[^{};]+override) 130126 
```
I don't see the 2nd printf inside that while-loop executing at all (even after 4-5 hours). Sure, this laptop is not the top-of-the-line config/fully loaded, but it does run quite a few EDA tools reasonably well. 

For experimentation sake, I emptied verilated.cpp and the issue shows up on next file. 

Any clues on how we can fix it cleanly please?

wsnyder, 2023-04-05:
I suspect some perl bug in expansion of the wildcards that is only in your version of perl.

Perhaps you'd be willing to rewrite it to instead anchor all of the patterns to start with \n?  E.g. "qr/\n[^\/]*virtual...."

Then

    while ($buffer =~ s/$pattern//sm) {

Be sure to temporarily "break" one of the .cpp files to make sure the regexps still detect failures (we don't have tests of the tests ;)

svenka3, 2023-04-11:
I did a fix as follows:

```perl
sub checkPattern {

    open(my $in_fileh, $filename)
      or die "Could not open file '$filename' $!";
    my @lines_l = <$in_fileh>;
    my $lineno = 0;

    foreach my $row (@lines_l) {
      $lineno++;
      if ($row =~ s/.*?^($pattern)//sm) {
        error("$filename:$lineno: $message");
      }
    }
```
Basically I am reading the contents to an array and looping through each line. This approach also simplifies the lineno detection I believe (no need of sub offset_to_lineno).

Is this approach fine? 

Thanks

svenka3, 2023-04-11:
If we care about pattern spread across few lines, perhaps the above can be enhanced to read few lines, concat them to a string and search - sort of in between each-line vs full-file-content.

wsnyder, 2023-04-11:
Yes, one of the patterns needs multiple lines.  I'd think looking across last 3 lines is sufficient.

wsnyder, 2023-05-27:
@svenka3 perhaps you'd be willing to finish this off and make a pull request?

svenka3, 2023-05-28:
Yes, sorry for the delay. Will work on it next week.

Thanks

svenka3, 2023-06-21:
OK, below is an attempt to look at 3 lines:

```perl
sub checkPattern {
    my $filename = shift;
    my $contents = shift;
    my $pattern = shift;
    my $message = shift;

    open(my $in_fileh, $filename)
      or die "Could not open file '$filename' $!";
    my @lines_l = <$in_fileh>;
    my $lineno = 0;

    my $num_rows = scalar(@lines_l);
    my @n_rows;

    for( $row_i = 0; $row_i < $num_rows; $row_i++) {
      $lineno++;
      $start_row = $row_i;
      $end_row = $start_row + 3;
      if ($end_row >= $num_rows) {
        $end_row = $num_rows - 1;
      }
      @n_rows = @lines_l [$start_row..$end_row];
      $buffer = join ("", @n_rows);
      if ($buffer =~ s/.*?^($pattern)//sm) {
        error("$filename:$lineno: $message");
      }
    }

    #my $offset = 0;
    #my $buffer = $contents;
    #while ($buffer =~ s/.*?^($pattern)//sm) {
    #  my $lineno = offset_to_lineno($contents, $offset + $-[-1]);
    #  $offset += $+[1];
    #  error("$filename:$lineno: $message");
    #}
}

```
I need to fix the $lineno - but wanted to run by your eyes @wsnyder - please review above. Idea is to grab 3 lines to a new array named n_rows and join them to a single buffer and run the pattern check.

wsnyder, 2023-06-25:
Seems reasonable, please turn it into a pull request, make the spacing 4 spaces per indent, put "my" in front of local variables, though I think you can just substitute in start_row and @n_rows.

closed 2023-09-16 by wsnyder