verilator/verilator #4086: Local variable access in fork..join_none - recent breakage

svenka3, 2023-04-04:
Hello,
  The SVA + SVUnit + Verilator code that worked for last few weeks seems now broken with an error as below (in latest build)

```log
%Error-UNSUPPORTED: ../src/AXI/axi4_lite//test_a_af_axi_errm_awburst.sv:47:48: Unsupported: variable local to a forking process accessed in a fork..join_any or fork..join_none
   47 |             if (svunit_ut.get_error_count() == local_error_count) begin
      |                                                ^~~~~~~~~~~~~~~~~
```

Error reported from:
```sh
Verilator 5.009 devel rev UNKNOWN.REV (mod)
```
Previous build where it worked:
```log
Verilator 5.009 devel rev v5.008-91-g0b96789e6
```
Should I assume the latest error is indeed true and hence I need to update SVUnit code? 

Thanks

wsnyder, 2023-04-05:
@kbieganski

kbieganski, 2023-04-05:
Is this variable local to the method that contains the fork? Then it's probably a correct error. That's because we can't guarantee the lifetime of local variables if the fork outlives the parent process.

Adding full support for that is difficult, but there is some low hanging fruit that can easily be implemented. Well, relatively easily. For example, using locals up until the first await in a fork is fine, because at that point the parent process is guaranteed to still exist. Another one is read-only access to locals after the fork point. In that case they can simply be copied to the forked processes.

kboronski-ant, 2023-10-11:
Does this issue still persist?

https://github.com/verilator/verilator/pull/4356 should have made it possible to refer to local variables even inside of `fork..join_none` and `fork..join_any` blocks.

wsnyder, 2023-10-21:
Closing assuming this is fixed as commented above, @svenka3 please comment and reopen if not.

closed 2023-10-21 by wsnyder