verilator/verilator #4087: Verilator wrongly complains about a syntax error with localparam

maestrosoft, 2023-04-05:
I am trying to "Verilator" the project Microsoft Zipline [at:](https://github.com/opencomputeproject/Project-Zipline).  Within the directory /Project-Zipline/rtl/cr_cceip_64 I run Verilator with the command:

`verilator --cc --exe --build -y ../cr_crcgc -y ../cr_prefix -y ../cr_prefix_attach -y ../cr_isf -y ../cr_osf -y ../cr_cg -y ../cr_huf_comp -y ../cr_lz77_comp -y ../cr_xp10_decomp -y ../cr_cceip_64_support -y ../common/nx_library -y ../common/cr_tlvp -y ../common -y ../common/axi -y ../common/cr_tlv_monitor -y ../common/include -y ../cr_cceip_64_sa -y ../cr_su -y ../mem_wrappers/bimc_master/top -j 0 -Wall cr_cceip_64.v`

Verilator exit with a syntax error:

%Error: ../common/nx_library/nx_ram_1r1w_indirect_access.v:110:24: syntax error, unexpected ':', expecting ',' or '}'
  110 |      =  { init_inc     : (N_INIT_INC_BITS>0)? TRUE : FALSE,
      |                        ^
        ../cr_prefix/cr_prefix_regfile.v:481:1: ... note: In file included from cr_prefix_regfile.v
        ../cr_prefix/cr_prefix_regfile.v:31:4: ... note: In file included from cr_prefix_regfile.v
        ../cr_prefix/cr_prefix.v:244:1: ... note: In file included from cr_prefix.v
        ../cr_prefix/cr_prefix.vh:33:1: ... note: In file included from cr_prefix.vh
        ../cr_prefix/cr_prefix.v:31:1: ... note: In file included from cr_prefix.v
        cr_cceip_64.v:517:1: ... note: In file included from cr_cceip_64.v
%Error: Exiting due to 1 error(s), 56 warning(s)

I checked the syntax of localparam instantation initialization:

   localparam capabilities_t capabilities_t_set
     =  { init_inc     : (N_INIT_INC_BITS>0)? TRUE : FALSE, 
          compare      : FALSE,
          reserved_op  : 4'b0,
          default      : TRUE};   

Everything seems fine and Synopsys doesn't complain when synthesizing this code. 

Is this a known bug of Verilator?

wsnyder, 2023-04-05:
I believe the code is attempting a pattern assignment, my reading of IEEE is this must use '{ not {.  I don't know why VCS would take it.  Does it work with '{?

maestrosoft, 2023-04-05:
Yes, it works with '{ . Why VCS would accept it might be related to some hidden details within the specification. I will try to find an answer and keep posted on the thread.

wsnyder, 2023-04-10:
Those websites are mentioning array initialization, not pattern assignments like your example code is. A IEEE reference is what is needed.

gbonneau-hardent, 2023-04-10:
You might be right. More googling showed me that VCS doesn't always follow the specification "by the book".  VCS might be attempting to "interpret" the context. Thus not giving an error.

... Because VCS is not following the rules of the LRM. (See section 10.10 of the 1800-2012 LRM). And in 10.10.1, there is an explicit example showing a similar case that is illegal...

https://verificationacademy.com/forums/ovm/how-assign-data-unpacked-structure

wsnyder, 2023-04-10:
I'm assuming that this isn't in IEEE so closing, but feel free to follow up otherwise.

closed 2023-04-10 by wsnyder