verilator/verilator #4090: The generated code fails to compile when two wires with bit selection are merged

yurivict, 2023-04-07:
Testcase:
```
module our(y, wire1, wire2);
        input wire wire1[3:3];
        input wire wire2[2:2];
        output wire y[1:0];

        assign y = {wire1, wire2};
endmodule
```

Error:
```
In file included from Vour__ALL.cpp:5:
./Vour___024root__DepSet_hd6c43f43__0.cpp:14:29: error: assigning to 'CData' (aka 'unsigned char') from incompatible type 'CData[1]' (aka 'unsigned char[1]')
    vlSelf->y[1U] = vlSelf->wire1;
                    ~~~~~~~~^~~~~
./Vour___024root__DepSet_hd6c43f43__0.cpp:15:29: error: assigning to 'CData' (aka 'unsigned char') from incompatible type 'CData[1]' (aka 'unsigned char[1]')
    vlSelf->y[0U] = vlSelf->wire2;
                    ~~~~~~~~^~~~~
2 errors generated.
```

Verilator should either print some errors, or succeed to build the code. The generated C++ code shouldn't just fail.

Version: 5.008
clang-14
FreeBSD 13.1

wsnyder, 2023-04-07:
>assign y = {wire1, wire2};

Verilator doesn't seem to currently known that wire1 which is a 1D unpacked array of a single entry can be implicitly sliced into a two-entry 1D array.  As a workaround you can use

>assign y = {wire1[3], wire2]2]};

Perhaps you could make a pull request to fix this (the problem is likely either in V3Width or in V3Slice, the .tree files should show if the data types are correct or not - if not it's the former).

wsnyder, 2025-09-09:
This works now in master, there have been multiple array handling fixes.

closed 2025-09-09 by wsnyder