verilator/verilator #4091: question: $finish behaviour

jotego, 2023-04-07:
`$finish` must be issued twice to stop the simulation. Why is that? Shouldn't the sim stop after a single `$finish`?

Another problem related to `$finish` is that it seems to bypass the C++ test bench in which the verilated code lives and the program exits without an opportunity to close the VerilatedVcdC object, so the waveforms around the event that caused the `$finish` are lost.

I am not including any test bench because this seems to be a deliberated behaviour so I may be missing something about the designer's intention here.

wsnyder, 2023-04-07:
$finish sets a flag your C++ code (or the default --main) should check before the next eval().  The default vl_finish routine in verilated.cpp special cases two $finish in the same timestep to exit.  This is somewhat historical.

Ideally your code would call $finish once then the next timestamp properly exit.

If you use one $finish and wait for eval() as described above your main code should close waves.  If you use two, the trace is supposed to have a exit handler registered which will clean up when vl_finish calls runFlushCallbacks.

Closing as answered, feel free to follow up or reopen if you have a --binary program or similar that doesn't flush waves on double-finish please post a bug (and/or ideally propose a pull to fix it).

wsnyder, 2023-04-07:
P.S. Note you can also provide your own vl_finish that does whatever you prefer.

closed 2023-04-07 by wsnyder

jotego, 2023-04-09:
Thank you!

With your hints and by digging in the _verilator_ source files I have found that I can call `uut.contextp()->gotFinish()` where `uut` is my verilog model.

I think that will sort me out :-)