verilator/verilator #4093: Verilated files fail to compile due to duplicate functions in 024root file

genfrontier, 2023-04-07:
Hey guys,
Trying to migrate from verilator 4.040 to 5.008, updated our build scripts and i am ussing gcc 10.4. building with -std=gnu++20 on fedora linux Here is the options used: 

--bbox-unsup
-Wall
-Wno-fatal
-Wno-style
-Wno-WIDTH
-Wno-UNUSED
-Wno-BLKSEQ
-Wno-PROCASSWIRE
-Wno-PINMISSING
-Wno-lint
--cc
--trace
--trace-max-array 512 
--Mdir verilated-hw/out
--prefix c_verilated_hw
--unroll-count 128 
--autoflush
--timing
--debug
--x-assign unique
--x-initial unique
${tensix-verilog-defs}
-f ${PROJECT_SOURCE_DIR}/common.f
--top-module top

This is a BIG project, after the verilate, the c++ errors look like this:

In file included from c_verilated_hw___024root__DepSet_h1d8360f1__0.cpp:8:                                                                                                                                                                                                                                                                                
c_verilated_hw___024root.h:883:5: error: 'CData c_verilated_hw___024root::<unnamed struct>::__Vtrigrprev__TOP__top_ten__inst_th0__inst_iss__td____PVT__pack_gated_clk' conflicts with a previous declaration                                                                                                      
  883 |     };                                                                                                                                                                                                                                                                                                                                            
      |     ^                                                                                                                                                                                                                                                                                                                                             
c_verilated_hw___024root.h:541:22: note: previous declaration 'CData c_verilated_hw___024root::<unnamed struct>::__Vtrigrprev__TOP__top_ten__inst_th0__inst_iss__td____PVT__pack_gated_clk'                                                                                                                       
  541 |         CData/*0:0*/ __Vtrigrprev__TOP__top__ten__inst_th0__inst_iss__td____PVT__pack_gated_clk;                                                                                                                                                                                                           
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                                                                                            
make[1]: *** [/poj/verilator/include/verilated.mk:239: c_verilated_hw___024root__DepSet_h1d8360f1__0.o] Error 1                                                                                                                                                                                 
make[1]: *** Waiting for unfinished jobs....

Has anyone gotten an idea what could be happening here? i am missing something ? i went thru the examples, and notes but can't find anything else needed.

Thanks

wsnyder, 2023-04-08:
A bug that might cause this was fixed just yesterday so please make sure you are using the latest.  Beyond that please create a small test case, hints might come from the signals it is showing, and/or ideally debug further, thanks.

genfrontier, 2023-04-10:
Thanks ! i updated to devel this morning and no longer see the issue, i'll close.

closed 2023-04-10 by genfrontier