verilator/verilator #4096: Using result of `$fopen` as initialization in variable declaration causes compile error

tudortimi, 2023-04-09:
I have the following code:

```
// file `test.sv`

module test;

  initial
    do_stuff();

  function automatic do_stuff();
    int f = $fopen("some_file", "w");
  endfunction

endmodule
```

I'm compiling it with `verilator --binary test.sv` and getting:

```
%Error: test.sv:8:13: syntax error, unexpected $fopen, expecting TYPE-IDENTIFIER
    8 |     int f = $fopen("some_file", "w");
      |             ^~~~~~
%Error: Exiting due to 1 error(s)
```

The code is legal and it works in commercial tools.

tudortimi, 2023-04-09:
I'm using version 5.008.

wsnyder, 2023-04-09:
This should be already fixed in master, please try that.

closed 2023-04-09 by wsnyder

tudortimi, 2023-04-09:
Sorry I didn't try with the latest version. It takes a long time to compile on my machine :( I scanned through the commits to get a feel of what was fixed since March, but I missed this.