verilator/verilator #4097: Using enclosing class name as function argument name leads to compile error

tudortimi, 2023-04-09:
I have a case where a function argument has the same name as a function argument:

```
file `test.sv`

class foo;

  function void some_function(int foo);
  endfunction

endclass
```

When I compile this with `verilator --binary test.sv`, I get:

```
%Error: test.sv:3:35: syntax error, unexpected TYPE-IDENTIFIER, expecting IDENTIFIER or randomize
    3 |   function void some_function(int foo);
      |                                   ^~~
%Error: Cannot continue
```

This works in a commercial tool.

tudortimi, 2023-04-09:
I'm using version 5.008.

wsnyder, 2023-04-09:
This is another case of #2412.  Due to complexities this is unlikely to be fixed soon, you will need to work around the naming.

closed 2023-04-09 by wsnyder

tudortimi, 2023-04-09:
I fixed the naming so this doesn't happen. In any case, the original issue shows some UVM code. Does this mean UVM support is a long way out?

wsnyder, 2023-04-09:
The https://github.com/chipsalliance/uvm-verilator repository has patched UVM to work around this, and the fix was pushed upstream to Accelera for the eventual new version of UVM.