verilator/verilator #4098: Extremely specific situaton leads to compile error on assignement of `const` variable in constructor

tudortimi, 2023-04-09:
I have a class with a `const` variable assigned in the constructor, but before that assignment there is another statement:

```
file `test.sv`

class foo;

  const string s;

  function new();
    int dummy = some_func();
    s = "foo";
  endfunction

  function int some_func();
  endfunction

endclass
```

When I call `verilator --binary test.sv`, I get:

```
%Error: test.sv:8:5: Assigning to const variable: 's'
    8 |     s = "foo";
      |     ^
%Error: Exiting due to 1 error(s)
```

`const` variables generally work, as removing the line with `int dummy` causes the code to compile. It's just this statement that causes the compile error.

tudortimi, 2023-04-09:
I'm using version 5.008.

tudortimi, 2023-04-09:
The code looks wonky, but that's only because it's focused on reproducing the issue. In the production code, I'm using a similar pattern, where I compute a couple of variables based on the constructor arguments and then assign some `const` variables based on them.

wsnyder, 2023-04-09:
Recursing into some_func() lost state related to being in new().

Just fixed it in master.

closed 2023-04-09 by wsnyder