verilator/verilator #4100: A way to add -isystem include to verilator build from source?

PeterMonsson, 2023-04-10:
I am trying to build verilator from source on a system with a strange FlexLexer.h location, but I can't find a way to make it build.

I have tried 
autoconf -I/home/petermonsson/miniconda3/include
./configure CFLAGS="-isystem /home/petermonsson/miniconda3/include"
CFLAGS="-isystem ../miniconda3/include" make -j

but none of them seem to work.

I don't see any effect of CFLAGS. When I look at the Makefiles I don't see CFLAGS as an option.

I would greatly appreciate any suggestions on how to proceed?

PeterMonsson, 2023-04-10:
Solved by copying FlexLexer.h to the ./include directory

closed 2023-04-10 by PeterMonsson

wallento, 2023-04-10:
It is still weird. I just checked configure with a bogus CFLAG and the check fails correctly. Setting isystem works and I can find it in src/Makefile_obj.