verilator/verilator #4101: Internal Error: V3DfgPeephole.cpp:1135: Mismatched RHS width

pdonahue-ventana, 2023-04-10:
When casting two things to int and adding them, it seems to sometimes complain that the sizes of the operands don't match.  The following code works on v4.210 but fails on v5.008:
```
module v(input logic t);
endmodule

module top(input logic [2:0] c);
  v v1((int'(c) + int'($countones(c))) > 2);
endmodule
```

```
$ verilator -cc top.sv
%Error: Internal Error: top.sv:5:17: ../V3DfgPeephole.cpp:1135: Mismatched RHS width
    5 |   v v1((int'(c) + int'($countones(c))) > 2);
      |                 ^
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

It seems to work if I change the v module's port to int and remove the `> 2`.

CentOS 7.

gezalore, 2023-04-10:
Thanks for the report. `-fno-dfg-peephole` to work around, otherwise I will fix it when I have a bit of time (not very soon)

gezalore, 2023-04-23:
This is caused by a change in #3725 that is not reflected in the DfgToAst code, I will post the fix shortly.

PR #4143 (2023-04-23): Fix DfgToAst conversion of CountBits (#4101)
Fixes #4101
closed 2023-04-23 by wsnyder