verilator/verilator #4102: Class with function that returns local struct type leads to internal error

tudortimi, 2023-04-10:
I have the following code:

```
// file `test.sv`

class some_class;

  typedef struct {
    string some_field;
  } some_struct;

  function some_struct some_func();
  endfunction

endclass
```

When I compile this with `verilator --binary test.sv` I get:

```
In file included from Vtest__Syms.h:19,
                 from Vtest.cpp:5,
                 from Vtest__ALL.cpp:3:
Vtest___024unit__03a__03asome_class__Vclpkg.h:27:10: fatal error: Vtest___024unit__03a__03asome_class.h: No such file or directory
   27 | #include "Vtest___024unit__03a__03asome_class.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

I played around with it a bit. If I move the typedef outside of the class it works. If I remove the function, it also works. It seems that it's the intersection of these two things that causes the issue.

tudortimi, 2023-04-10:
I'm using the latest Verilator (v5.008-121-g2ab34b5ee),

wsnyder, 2023-04-10:
Fixed, thanks for the good test case.

closed 2023-04-10 by wsnyder

tudortimi, 2023-04-11:
Also thanks for fixing this so quickly.