verilator/verilator #4108: Support for std::bitset instead of CData, SData, IData, QData, and VlWide.

phelter, 2023-04-11:
What would you like added/supported?

Looks like in v5.00+ there is a requirement for C++20.  With this in mind Would it be possible to identify whether the underlying data types generated/used for all signals could be converted to `std::bitset` instead of 4 different data types?

Pros:
* Provide a consistent and single type class for signals and ports etc. 
* stricter type definition - eg. a VL_SIG8(a, 3, 0) - won't be type equivalent to a VL_SIG8(a, 4, 0).
* Can pass around lvalue  and be able to identify the size of the bitvector directly from the type
* Able to operate on all bits in a bitset as a single element - no matter the size - no more looping through the words.
  * Most verilog binary operators supported in std::bitset (AND, OR, XOR, NOT, SHIFT_LEFT, SHIFT_RIGHT).
* string representations already present `std::to_string`

Cons:
* Slightly larger memory storage for anything less than 64-bits (on a 64-bit machine).
* Might have implications with support for std::bitset for various compilers
  *  if not supported could implement a std compilant one and use that if not supported directly.
* Most likely have implications on how users consume the signals and ports. - Rework for testbenches upgrading to this version.

Or are there other reasons for not using bitset that I'm unaware of?

wsnyder, 2023-04-11:
- C++20 is not required, it is optional, C++20 is needed only for thread support.
- Requiring all 64 bits would reduce performance by ~20% due to cache pressure alone, perhaps more in total.
- V3Expand and other optimizations that are at the word level could not run, another 10% loss or so.
- Based on old data C++ compile times would go up by 2x (might have improved, I haven't tried in a while)

So, I appreciate the question, but I don't see this being a benefit to end users, so closing, but feel free to follow up.

BTW a related question might be if Verilated should internally use 64 bit words rather than 32 bit for word operations. I tried this and due to cache pressure it was harmful overall.

closed 2023-04-11 by wsnyder

phelter, 2023-04-13:
Thanks for the response.

An alternative is to have all ports in one big `std::bit_set` and then you wouldn't waste memory space for a whole bunch of single bit input output values - currently each is 8 bits minimum for a single bit, but with this approach you could kill 2 birds with 1 stone:

* Reduce small number of bits.
* Parameterize the interfaces - with:

```verilog
parameter int X_SIZE = 8, 
parameter int Y_SIZE = 16
```
<template size_t X_SIZE = 8, size_t Y_SIZE = 16>
```
And reduce the number files that get written out.  the X_SIZE and Y_SIZE would then be an input into the bit_set()

It would be nice if each instantiation wasn't it's own verilated model which would allow you to include verilated sub-modules inside a verilated module without having to generate for the specific parameters.