verilator/verilator #4109: Internal error when using array query function on list of objects inside begin...end block within function

tudortimi, 2023-04-11:
I have the following code:

```
// file test.sv

class some_class;

  function int get_val();
  endfunction

endclass

function void do_stuff();
  some_class list_of_some_class[];
  
  // might have other statements above this line, so we need the begin...end to declare a new
  // throwaway variable
  begin
    some_class match[$] = list_of_some_class.find() with (item.get_val() == 1);
  end
endfunction
```

Compiling it with `verilator --binary test.sv` leads to:

```
In file included from Vtest__ALL.cpp:14:
Vtest___024unit__DepSet_h0b4ff0d4__0__Slow.cpp: In function ‘void Vtest___024unit___eval_static__TOP____024unit(Vtest___024unit*)’:
Vtest___024unit__DepSet_h0b4ff0d4__0__Slow.cpp:18:19: error: ‘item’ was not declared in this scope; did you mean ‘tm’?
   18 |     VL_NULL_CHECK(item, "test.sv", 15)->__VnoInFunc_get_val(vlSymsp, __Vtask_get_val__0__Vfuncout);
      |                   ^~~~
      |                   tm
```

This only seems to happen when using an array of objects; an array of int won't cause this issue. The `begin...end` block is also important. Also, if I declare the function as `function automatic void do_stuff()`, then it doesn't crash. Using `function static void do_stuff()` or declaring the function under a class causes it to crash with the same error message.

tudortimi, 2023-04-11:
I'm using the latest Verilator, v5.008-125-g0307d59c1.

tudortimi, 2023-04-11:
In the production code, I use the `begin...end` block to declare a throwaway variable. It's something like this:

```
function void svunit_testsuite::report();
  int  pass_cnt;
 
  foreach(list_of_testcases[i])
    list_of_testcases[i].report();

  begin
    svunit_testcase match[$] = list_of_testcases.find() with (item.get_results() == PASS);
    pass_cnt = match.size();
  end

  // ...
endfunction
```

This is a pretty common idiom.

tudortimi, 2023-04-11:
I'm thinking of extracting a function out of the `begin...end` block, as it would clean up the code a bit, so this doesn't really affect me. I found the crash interesting, though, and fixing it might also fix other hidden issues.

tudortimi, 2023-04-11:
As a workaround I can also just not use the array query function and use regular `foreach` loop.

wsnyder, 2023-04-12:
Duplicate of #4003 I presume.

tudortimi, 2023-04-12:
It's exactly that, lol. It seemed so specific that I didn't even think to look through the existing issues.