verilator/verilator #4117: Assignment of wire to other portions of itself creates a VGate.cpp:1004 Internal Error

phelter, 2023-04-13:
Using a third-party processor that is from a company that is also an `upper body appendage`, so cannot provide you a code drop.

Version: Verilator 5.006 2023-01-22 rev v5.006 on Ubuntu-22.04

When verilating a particular portion of the verilog - I get the following errors:

```
%Error: Internal Error: <upper body appendage file>.vline:loc: ../V3Gate.cpp:1004: Can't replace lvalue assignments with const var
```

I originally added -fno-gate, but have been seeing some weird things that are going on in the simulation output vcd files, so decided to fix this and see if those resolve.

*Note:* Would be nice to allow the -fno-gate to be applied to only a single file of a larger design.

I tracked it down to the following style of code - and the driver of the wire - names of variables/wires have been changed.

wire [4:0] foo;
`ifdef VERILATOR_WORKAROUND

  wire [4:1] foo_order;
  assign foo_order[1] = foo[0];
  assign foo_order[2] = |foo[1:0];
  assign foo_order[3] = |foo[2:0];
  assign foo_order[4] = |foo[3:0];

  assign foo[0] = a_bar_status[1];
  assign foo[1] = b_bar_status[1] & ~foo_order[1];
  assign foo[2] = c_bar_status[1] & ~foo_order[2];
  assign foo[3] = d_bar_status[1] & ~foo_order[3];
  assign foo[4] = e_bar_status[1] & ~foo_order[4];
`else
  assign foo[0] = a_bar_status[1];
  assign foo[1] = b_bar_status[1] & ~foo[0];
  assign foo[2] = c_bar_status[1] & ~(|foo[1:0]);
  assign foo[3] = d_bar_status[1] & ~(|foo[2:0]);
  assign foo[4] = e_bar_status[1] & ~(|foo[3:0]);
`endif
```

Since the above with `-DVERILATOR_WORKAROUND=1` works and there are no other internal errors, I'm no longer investigating this issue, but I thought I'd add it here so that you can augment your test suite to include this style and possibly fix it in the future.

wsnyder, 2023-04-13:
Could you make this into a self-contained test I can run directly that causes the error?

There are problems with analyzing clocks with -fno-gate, and generally this was hard to fix, and there aren't often errors with that optimization, so generally I'd avoid -fno-gate.

phelter, 2023-04-28:
I attempted to create a simple version of this scenario, but the simpler version detects:
`%Warning-UNOPTFLAT` for the signal in question `foo` when the smaller one is built, but this warning is not detected for the full implementation version.

The code is practically identical with the exception of the fan-in to the foo signals.

I've added lint_off around that to allow it to generate.

Here's the example:
[issue_4117_hdl.tar.gz](https://github.com/verilator/verilator/files/11348892/issue_4117_hdl.tar.gz)

I'll let you verilate it the way you want.
* I used cmake and my own verilator script to separate the `verilator_base` library code from the rest and consume that as a dependency to the actual verilated code.  The hdl is also defined as native CMake Language SystemVerilog/Verilog. This allows me to build up the dependency tree of the verilog code independent of the verilated code, and perform the generation of verilated code at the Build stage and not the Configuration stage of CMake. And removes the need for using the generated CMake Files all together.
* But I'm providing the verilate command output from ninja.  
  * mkdi

```bash
mkdir verilator_debug_issues/build
mkdir verilator_debug_issues/issue_4117/
cd verilator_debug_issues/issue_4117
tar -xzvf issue_4117_hdl.tar.gz .

verilator_bin --compiler clang --prefix Vissue_4117_example --Mdir <root>/verilator_debug_issues/build/issue_4117/hdl/CMakeFiles/issue_4117_hdl.dir/Vissue_4117_example.dir --make cmake -Wall -Wpedantic --assert -sv --hierarchical --trace --trace-structs --coverage -DVERILATOR_WORKAROUND=1 --top issue_4117_example --cc <root>/verilator_debug_issues/issue_4117/hdl/issue_4117_example.v <root>/verilator_debug_issues/issue_4117/hdl/status_gen.v
```
* And when not using the WORKAROUND just remove `-DVERILATOR_WORKAROUND=1`

Again - please be aware that this does *NOT* reproduce the V3Gate.cpp issue with the smaller code instance.  But it might point you in the right direction - especially if the UNOPTFLAT is not detected in the larger case.

phelter, 2023-04-28:
So I've found the fix for this @wsnyder .

The problem is the the small test detects the UNOPTFLAT.  The larger test case with more files did not.  Do not know why - it was not disabled in any way.

```verilog
  assign foo[0] = a_bar_status[1];
  assign foo[1] = b_bar_status[1] & ~foo[0];
  assign foo[2] = c_bar_status[1] & ~(|foo[1:0]);
  assign foo[3] = d_bar_status[1] & ~(|foo[2:0]);
  assign foo[4] = e_bar_status[1] & ~(|foo[3:0]);
```
When I removed the optimization and added in the configuration file:
`split_var -module "module" -var "foo"` - the `V3Gate` does not occur.  So there is a bug here in that the above code doesn't always detect UNOPTFLAT, but the good news is it gets detected when attempting to perform the Optimization `V3Gate` .

Might want to update the documentation in a way to describe this until the UNOPTFLAT detection is fixed.

wsnyder, 2023-04-29:
I would turn off UNOPTFLAT  (-Wno-UNOPTFLAT) and see if that then shows the error. I would expect UNOPTFLAT in your example, but do not know why it didn't show up normally.

Anyhow, think I still need an example showing the V3Gate problem to debug further, the snippit (without UNOPTFLAT) is ok, so some upstream or downstream logic is also needed to hit the bug.

phelter, 2023-05-01:
Yeah I know I'll try and get that - there's a very very large fan-in so it is not very feasible.  But will attempt.

wsnyder, 2023-07-02:
Closing until get example.

closed 2023-07-02 by wsnyder