verilator/verilator #4120: Verilation issues with virtual tasks when inherting across packages

GregAC, 2023-04-14:
When a class inherits from another class that sits in a different package and the inheriting class overrides a virtual task that does nothing in the base class but waits in the derived class a verilation issue can occur. The generated C++ returns a `VlCoroutine` in the derived class implementation but not in the base class for the overridden task leading to a C++ compile error.

This is seen in the following example code:

```systemverilog
package test_pkg;

class base;
  virtual task foo();
  endtask
endclass

endpackage

import test_pkg::*;

class derived extends base;
  int field;

  virtual task foo();
    wait (field == 0);
  endtask
endclass

module test_virtual_test;
  initial begin
    derived derived_inst = new;
    derived_inst.field = 1;
    derived_inst.foo();
  end
endmodule

```

When built with (`class_bug.sv` contains the above example)

```bash
verilator --binary class_bug.sv
```
Resulting in the following compile error:

```
In file included from Vclass_bug__Syms.h:20,
                 from Vclass_bug.cpp:5,
                 from Vclass_bug__ALL.cpp:3:
Vclass_bug___024unit__03a__03aderived__Vclpkg.h:39:25: error: conflicting return type specified for ‘virtual VlCoroutine Vclass_bug___024unit__03a__03aderived::__VnoInFunc_foo(Vclass_bug__Syms*)’
   39 |     virtual VlCoroutine __VnoInFunc_foo(Vclass_bug__Syms* __restrict vlSymsp);
      |                         ^~~~~~~~~~~~~~~
In file included from Vclass_bug___024unit__03a__03aderived__Vclpkg.h:28,
                 from Vclass_bug__Syms.h:20,
                 from Vclass_bug.cpp:5,
                 from Vclass_bug__ALL.cpp:3:
Vclass_bug_test_pkg__03a__03abase__Vclpkg.h:33:18: note: overridden function is ‘virtual void Vclass_bug_test_pkg__03a__03abase::__VnoInFunc_foo(Vclass_bug__Syms*)’
   33 |     virtual void __VnoInFunc_foo(Vclass_bug__Syms* __restrict vlSymsp);
      |                  ^~~~~~~~~~~~~~~
make: *** [/tools/verilator/latest/share/verilator/include/verilated.mk:239: Vclass_bug__ALL.o] Error 1
```

This has been seen on the latest master (`Verilator 5.009 devel rev v5.008-130-g23f90b13f`)

The problem goes away when `base` and `derived` are in the same package.

GregAC, 2023-04-14:
@wsnyder apologies didn't select a template so this failed to get labeled 'new' as it should have been.

wsnyder, 2023-05-03:
@kbieganski perhaps you can take a look or if @GregAC is willing suggest what to do to propose a pull to fix it?

kbieganski, 2023-05-04:
#4169 should fix it. Thanks for the minimal test case @GregAC!

PR #4169 (2023-05-04): Fix marking overridden methods as coroutines (#4120)
This patch fixes two cases where methods in base classes were not being marked as coroutines, even though they were being overridden by coroutines.
- One case is the class member cache not getting refreshed for searched classes.
- The other is when the overriding methods are not declared as `virtual`. In that case, the `isVirtual()` getter on such a method returns false, which led to `V3Timing` skipping the step of searching for overridden methods.

Fixes #4120.
closed 2023-05-04 by kbieganski