verilator/verilator #4121: Acess to full pin name in simulation

Shatur, 2023-04-14:
In generated file `__024root.h` we have multiple `CData` declaration that store data. Is it possible to access to information to which cell and pin they correspond at the simulation stage?
Since we can dump FST or VCD, I think this information is accessible, but is it possible to access this information at runtime to do something with specific name?

wsnyder, 2023-04-14:
The VPI supports this in other simulators, see the IEEE vpiPort spec for details. This has not been implemented in Verilator, if you want to support it you'll need to contribute a patch.

Shatur, 2023-04-14:
@wsnyder Yes, I also thinking about VPI approach, but I afraid that this approach will be a bit slow. I could try to send a PR. But not sure where to start. How would you see this feature? Also Verilator somehow maps FST and VCD using AST, it's just not exported to users, right?

wsnyder, 2023-04-14:
Verilator creates the FST/VCD (or VPI) information from the internal AST tree directly, yes.   You could use --xml option if you just want the port information without needing to access the signal data.

Shatur, 2023-04-14:
> You could use --xml option if you just want the port information without needing to access the signal data.

This was my initial approach. I tried parsing XML and generate additional `.cpp` file that map names to variables. But my problem is that sometimes it optimizes some pins and one variable could correspond to multiple pins.
So I probably should parse `.tree` files, but where can I find information which variable correspond to which pin?

wsnyder, 2023-04-14:
Do not parse .tree files, absolutely no guarantees there on the format being stable.

Can you describe what you want, I'm confused by your wanting information post-optimization, and not wanting the VPI which is the typical solution for this.

Shatur, 2023-04-14:
Thanks for the advice, I probably should stick with VPI as you suggested.
I trying to simulate so-called fault injection by changing specific pin in the middle of the simulation. Am I understand correctly that Verilator optimizes a lot of things and makes my use-case not applicable?

wsnyder, 2023-04-14:
Adding faults at the output won't work generally due to optimizations, they should be injected at the front end typically. See #2676 for fault injection discussion.

Closing as I think is answered, but feel free to follow up.

Shatur, 2023-04-14:
Thanks for the info!

Shatur, 2023-04-17:
@wsnyder I found your old comment about Verilator:

> Verilator can simulate a gate level netist, though it may optimize the gates so aggressively and into such a different form as to be unsuitable to represent gate-level faults. Many of these optimizations cannot be disabled.

Is this still true? Does it mean that Verilator is not suitable for gate-level faults simulation?

wsnyder, 2023-04-17:
It is not suitable for finding timing dependent faults, or falults after Verilation. If you insert logic before Verilator, it can be used to find logical faults.

Shatur, 2023-04-17:
> If you insert logic before Verilator, it can be used to find logical faults.

Do you mean that I can insert a fault after starting simulation?

wsnyder, 2023-04-17:
Insert fault insertion logic in the Verilog code before Verilator. See the other issue please.

Shatur, 2023-04-17:
Got it, thank you!