verilator/verilator #4125: Add support for `disable fork`

tudortimi, 2023-04-15:
Currently, Verilator issues `Unsupported: disable fork statements`. Is support for `disable fork` something that would be reasonably easy to add? Would it also be do-able for a beginner?

tudortimi, 2023-04-15:
@PeterMonsson We absolutely cannot have SVUnit in Verilator without this, because `disable fork` is used to stop the tests when errors are encountered and to move on to the next test.

wsnyder, 2023-04-16:
@kbieganski  what do you think about difficulty?

As a first step, how about submitting a patch that splits t_fork_disable.v into also t_fork_join_none.v and t_fork_wait.v witout "join_none" and "wait_fork", then make t_fork_disable.v into a good self-checking test, including testing a disable with nothing pending. Also a t_fork_disable_bad.v which has one not under a fork at all (so is illegal).  Please check them against other simulators.

These will of course fail as unsupported errors, but we can commit them as "fails" tests so the tests are ready for the support.

kbieganski, 2023-04-17:
Right now it's pretty difficult for a beginner, at least in the general case. There's currently no way to stop processes from the outside. You would need to make it possible to locate a particular suspended coroutine in a queue and kill it.

However, we (@antmicro) are working on `std::process`, and we're getting pretty close. `std::process` has methods like `suspend()`, so part of that work is enabling external control of processes. I think `disable fork` is basically syntax sugar for `process::kill()` (not sure about its effect on `status()`). So once we land our changes in mainline, `disable fork` could be implemented by either using `process` directly, or using the same runtime infrastructure.

Unfortunately, at this point we are not planning on implementing `process::kill()`. But `suspend()` is definitely happening, and it's probably not that difficult to go from `suspend()` to `kill()`. Then it should be fairly easy to go from that to `disable fork`. Though I imagine some parts might be challenging:
- nested forks,
- lifetime of a local `process` variable (see #4086),
- possibly memory management?

I suggest keeping an eye out for an `std::process` PR from us in a few weeks (hopefully). Otherwise, if you want to try a different approach, I recommend reading the _Timing_ section of the internals doc. Or the entire doc, really – it's very helpful for new Verilator devs. Also, look for some C++ coroutine tutorials if you haven't had a chance to check them out. (I guess that knowledge will prove useful even if you decide to go with `process::kill`)

wsnyder, 2023-04-17:
@tudortimi you mentioned your main use case was finishing a test and moving to another.  If you know where the forked process(es) wait is occurring (e.g. some loop or poll on queue), perhaps you can introduce a "exitFork" variable into your SV code, or Verilator's code, and test that in the process?  This won't exit arbitrary loops etc but perhaps it gets most of what you want?

tudortimi, 2023-04-17:
@wsnyder Thanks for the suggestion. I'll have a look at implementing something like this.

tudortimi, 2023-05-08:
Is there any update on this? I'm more or less happy with the current state of Verilator and can merge my code to support it, but if `disable fork` would land in a couple of days, I would rather wait a bit longer.

kbieganski, 2023-05-08:
Note that we're not working on `disable fork` at this time. We're working on some infrastructure that should make it fairly easy to implement `disable fork`. So if it's urgent, I would advise to merge your workaround.

kbieganski, 2023-06-01:
https://github.com/verilator/verilator/pull/4212 has been merged. Thanks to this, one could implement `disable fork` by getting the `std::process` instance of each forked process and calling `kill` on them.

wsnyder, 2023-09-09:
Note "disable fork" is one of a few remaining blockers to UVM #1538

```
%Error-UNSUPPORTED: t/t_uvm_pkg_all.vh:11187:12: Unsupported: disable fork statements
%Error-UNSUPPORTED: t/t_uvm_pkg_all.vh:15165:5: Unsupported: disable fork statements
%Error-UNSUPPORTED: t/t_uvm_pkg_all.vh:19209:7: Unsupported: disable fork statements
```

PR #4569 (2023-10-16): Support `disable fork`
This adds `disable fork`.
For it to work, VlProcess objects now maintain information on hierarchy of running processes.
There was a need for another rework in VlProcess propagation, to cover a more general case in which the hierarchy matters.

Closes #4125.
closed 2023-10-16 by kiryk