verilator/verilator #4129: XML output cell hier of generate block

Risto97, 2023-04-17:
When having a module instantiation in generate block like below:
```
module submod();
    wire some_sig;
endmodule

module test(
);
    if(1) begin : some_gen
        submod submod_i();
    end
endmodule
```
I think that the generated XML file is incorrect, specifically <cells> section cells attribute hier.
This is the output I get:

```diff
<?xml version="1.0" ?>
<!-- DESCRIPTION: Verilator output: XML representation of netlist -->
<verilator_xml>
  <files>
    <file id="a" filename="&lt;built-in&gt;" language="1800-2017"/>
    <file id="b" filename="&lt;command-line&gt;" language="1800-2017"/>
    <file id="c" filename="test.v" language="1800-2017"/>
  </files>
  <module_files>
    <file id="c" filename="test.v" language="1800-2017"/>
  </module_files>
  <cells>
    <cell loc="c,5,8,5,12" name="test" submodname="test" hier="test">
-     <cell loc="c,9,12,9,20" name="submod_i" submodname="submod" hier="test.submod_i"/>
+     <cell loc="c,9,12,9,20" name="submod_i" submodname="submod" hier="test.some_gen.submod_i"/>
    </cell>
  </cells>
  <netlist>
    <module loc="c,5,8,5,12" name="test" origName="test" topModule="1">
      <begin loc="c,8,15,8,23" name="some_gen">
        <instance loc="c,9,12,9,20" name="submod_i" defName="submod" origName="submod_i"/>
      </begin>
    </module>
    <module loc="c,1,8,1,14" name="submod" origName="submod">
      <var loc="c,2,10,2,18" name="some_sig" dtype_id="1" vartype="logic" origName="some_sig"/>
    </module>
    <typetable loc="a,0,0,0,0">
      <basicdtype loc="c,2,10,2,18" id="1" name="logic"/>
    </typetable>
  </netlist>
</verilator_xml>

```
I would expect that the path to the submodule would also include some_gen. Am I correct to assume that?
This way I cannot access VPI signal based on this path, as `some_sig` signal for VPI  is `test.some_gen.submod_i.some_sig`

May we assist you in trying to fix this in Verilator yourself?
Yes, I could fix it if it is bug after all, and I can get some directions.

Risto97, 2023-04-18:
@wsnyder I tried to fix this in #4133 , further comments would be helpful

wsnyder, 2023-04-24:
Thanks for working on this, especially working on a first commit and keeping at it!

closed 2023-04-24 by wsnyder