verilator/verilator #4132: Verilator allows function outputs to be wires

Timmmm, 2023-04-18:
The following code compiles with no errors or warnings:

```
module main(
    output wire bit valid,
    output wire bit ready
);
    import "DPI-C" context function void example_dpi(output bit valid);

    function void example_sv(output bit ready);
        ready = 1;
    endfunction

    initial begin
        example_dpi(valid);
        example_sv(ready);
    end

endmodule
```

However it doesn't make any sense does it? A function output can't be connected to a `wire`; only a `var`. The LRM says:

> If the argument in the subroutine is declared as an output or an inout, then the corresponding expression
in the subroutine call shall be restricted to an expression that is valid on the left-hand side of a procedural
assignment.

And

> The left-hand side of a procedural assignment can take one of the following forms:
>  * Singular variables, as described in 6.4
>  * Aggregate variables, as described in Clause 7
>  * Bit-selects, part-selects, and slices of packed arrays
>  * Slices of unpacked arrays

I'm using Verilator 5.004 2022-12-14 rev v5.004.

wsnyder, 2023-04-18:
Yes, lint checking should be improved.  Perhaps you could contribute a pull to fix this?

fuad1502, 2024-03-12:
Hi @wsnyder 

I tried to fix this issue in [#4966 ](https://github.com/verilator/verilator/pull/4966). Can you help me review the changes?

By the way, this is my first time contributing to an open source project, so please let me know if there are certain conventions that I should've followed. For example, should I have asked first whether I can try to fix this issue before making the pull request?

Any guidance would be greatly appreciated 😊

closed 2024-03-16 by wsnyder