verilator/verilator #4135: Verilator use incorrect SystemC copy macro

Sustrak, 2023-04-19:
In a large code base I'm working on I found the following bug in the verilated C++ code. The faulty C++ line is the following:

```c++
VL_ASSIGN_SWW(91,vlSelf->d_rn_req_q_data2_o, vlSelf->d_rn_req_q_data_o);
```

The macro declaration is the following:
```c++
#define VL_ASSIGN_SWW(obits, svar, rwp) \
    { \
        sc_bv<(obits)> _bvtemp; \
        for (int i = 0; i < VL_WORDS_I(obits); ++i) _bvtemp.set_word(i, (rwp)[i]); \
        (svar).write(_bvtemp); \
    }
```

The declaration of `d_rn_req_q_data2_o` and `d_rn_req_q_data_o` are the following:
```c++
sc_out<sc_bv<91> > d_rn_req_q_data2_o;
...
sc_out<sc_bv<91> > d_rn_req_q_data_o;
```

I think the expected code, produced by verilator, should be:
```c++
d_rn_req_q_data2_o = d_rn_req_q_data_o; 
```
Since both variables have same type, and SystemC overloads the `=` operator.

If more info about the veriated code is needed I can provide it.

> What 'verilator --version' are you using?

`Verilator 5.009 devel rev v5.008-90-g947402bc5`

> May we assist you in trying to fix this in Verilator yourself?

Sure, I would kindly check if the proposed solutions fix this issue in my codebase

wsnyder, 2023-04-19:
That macro is intended to convert from internal to systemc format. It should not be in the created code passing in two systemc variables. Do you have a complete example?

wsnyder, 2023-04-20:
Please also give the sc_main that shows the compile error, as this looks OK to me visually, thanks.

Sustrak, 2023-04-20:
Hi, I'm not able to hit the compilation bug with this simple code, but, since both input signals and output signals are systemC data types, no conversion should be needed right? Hence the use of the macro `VL_ASSIGN_SWW` should be replaced, probably, by the `=` operator. Or I'm missing something?

wsnyder, 2023-04-20:
VL_ASSIGN_SWW(235,vlSelf->a2_o, __Vcellinp__test__a_i);

This is assigning to SystemC from internal.  vlSelf->a2_o is systemc, __Vcellinp__test_a_i is internal. So the code is correct I believe.

Sustrak, 2023-04-20:
RIght, I see it now. I'll try to came up with another example.

In the meantime is some print/modification in the verilator code I could do to give you more insight of the problem?

wsnyder, 2023-04-21:
If you want to trace it down to the C++ source and fix it, that will help determine also how to write a test.  I don't think any additional debug is needed as the hard part is the test to make sure it is fixed.

wsnyder, 2023-04-25:
visit(AstNodeAssign) in V3EmitCFunc.h - basically any AstAssign in the tree that is a SystemC variable.  There shouldn't be a SystemC variable on both sides of an assign.

Sustrak, 2023-04-26:
Hi @wsnyder I have managed to print the assign node and it looks like:

```
     ASSIGNW 0x121aef30 {cc116cf} @dt=0xc66ed90@(G/wu96/91)
    1:   VARREF 0x121aefe0 {cc116cf} @dt=0xc66ed90@(G/wu96/91)  d_rn_req_q_data_o [RV] <- VAR 0x32b28d0 {cc68cf} u4=0x3924260 @dt=0x3ab7a60@   (w91)ref  d_rn_req_q_data_o [SC] [PO] OUTPUT [P] PORT
    2:   VARREF 0x121af0f0 {cc116cf} @dt=0xc66ed90@(G/wu96/91)  d_rn_req_q_data2_o [LV] => VAR 0x590b9e0 {cc116cf} u4=0x3924260 @dt=0x3ac9ec0@ (w91)ref  d_rn_req_q_data2_o [SC] [PO] OUTPUT [P] PORT
-----------------------------
VL_ASSIGN_SWW(91,vlSelf->d_rn_req_q_data2_o, vlSelf->d_rn_req_q_data_o);
```

In `V3EmitCFunc.h:338` I have found where the `VL_ASSIGN_SWW` macro is being emited:
```c++
} else if (AstVar* const varp = AstVar::scVarRecurse(nodep->lhsp())) {
            putbs("VL_ASSIGN_");  // Set a systemC variable
            emitScIQW(varp);
            emitIQW(nodep);
            puts("(");
            puts(cvtToStr(nodep->widthMin()) + ",");
            iterateAndNextConstNull(nodep->lhsp());
            puts(", ");
        }
```

Since both `VARREF` are `SC`, the expected behavior is to not reach this part of code and branch somewhere else? Or are we missing a condition inside the `else if (AstVar* const varp = AstVar::scVarRecurse(nodep->lhsp()))`, to check the case where both `VARREF` are `SC`?

Thanks for the help

wsnyder, 2023-04-26:
There should not be a systemc variable on both sides of the assign, so trace the tree files back, likely some optimization is doing that and needs to be told not to do it on SystemC variables on both sides.

Sustrak, 2023-04-27:
Hi @wsnyder could you point out in which part of the code may I find this posible optimization/transformation of the AST? Also wich kind of node should we emit when in both sides there are SystemC variables?

wsnyder, 2023-04-27:
If I knew where it was happening to tell you I'd fix it ;)

There should NEVER be a SystemC variable on both sides.  You need to look at the .tree files to find where they are being made (that is an ASSIGN with two VARREFS with [SC] under them). See internals.rst for more information on tree files. 

I think it may be easier for you to try to reduce your test code to the smallest possible that shows this problem, make sure it shows no private information, and send the test case for others to debug.

Sustrak, 2023-05-04:
Hi @wsnyder sorry for the late response, but I'm not able to generate a smaller testcase that reproduces the error.

I've managed to get the tree outputs:

```
obj_dir/Vlc_th_001_cells.tree:    1:2: ASSIGNW 0x4dac780 <e303975#> {cb722bp} @dt=0@
obj_dir/Vlc_th_001_cells.tree-    1:2:1: PARSEREF 0x4dac6a0 <e303973#> {cb722br} @dt=0@  d_rn_req_q_data_o [TEXT]
obj_dir/Vlc_th_001_cells.tree-    1:2:2: PARSEREF 0x4dac4f0 <e303974#> {cb722ai} @dt=0@  d_rn_req_q_data2_o [TEXT]
--
obj_dir/Vlc_th_006_linkjump.tree:    1:2: ASSIGNW 0x4dac780 <e303975> {cb722bp} @dt=0@
obj_dir/Vlc_th_006_linkjump.tree-    1:2:1: VARREF 0x70dde70 <e505581#> {cb722br} @dt=0@  d_rn_req_q_data_o [RV] <- VAR 0x4cd7570 <e384500#> {cb68cf} @dt=0@  d_rn_req_q_data_o OUTPUT PORT
obj_dir/Vlc_th_006_linkjump.tree-    1:2:2: VARREF 0x70ddf80 <e505584#> {cb722ai} @dt=0@  d_rn_req_q_data2_o [LV] => VAR 0x4cedc70 <e384725#> {cb116cf} @dt=0@  d_rn_req_q_data2_o OUTPUT PORT
--
obj_dir/Vlc_th_007_linkinc.tree:    1:2: ASSIGNW 0x4dac780 <e303975> {cb722bp} @dt=0@
obj_dir/Vlc_th_007_linkinc.tree-    1:2:1: VARREF 0x70dde70 <e505581> {cb722br} @dt=0@  d_rn_req_q_data_o [RV] <- VAR 0x4cd7570 <e384500> {cb68cf} @dt=0@  d_rn_req_q_data_o OUTPUT PORT
obj_dir/Vlc_th_007_linkinc.tree-    1:2:2: VARREF 0x70ddf80 <e505584> {cb722ai} @dt=0@  d_rn_req_q_data2_o [LV] => VAR 0x4cedc70 <e384725> {cb116cf} @dt=0@  d_rn_req_q_data2_o OUTPUT PORT
--
obj_dir/Vlc_th_009_linkdotparam.tree:    1:2: ASSIGNW 0x4dac780 <e303975> {cb722bp} @dt=0@
obj_dir/Vlc_th_009_linkdotparam.tree-    1:2:1: VARREF 0x70dde70 <e505581> {cb722br} @dt=0@  d_rn_req_q_data_o [RV] <- VAR 0x4cd7570 <e384500> {cb68cf} @dt=0@  d_rn_req_q_data_o OUTPUT PORT
obj_dir/Vlc_th_009_linkdotparam.tree-    1:2:2: VARREF 0x70ddf80 <e505584> {cb722ai} @dt=0@  d_rn_req_q_data2_o [LV] => VAR 0x4cedc70 <e384725> {cb116cf} @dt=0@  d_rn_req_q_data2_o OUTPUT PORT
--
obj_dir/Vlc_th_011_width.tree:    1:2: ASSIGNW 0x4dac780 <e2311044#> {cb722bp} @dt=0x4cedb50@(w91)ref
obj_dir/Vlc_th_011_width.tree-    1:2:1: VARREF 0x70dde70 <e2311045#> {cb722br} @dt=0x4cd7450@(w91)ref  d_rn_req_q_data_o [RV] <- VAR 0x4cd7570 <e2307794#> {cb68cf} @dt=0x4cd7450@(w91)ref  d_rn_req_q_data_o OUTPUT PORT
obj_dir/Vlc_th_011_width.tree-    1:2:2: VARREF 0x70ddf80 <e2311043#> {cb722ai} @dt=0x4cedb50@(w91)ref  d_rn_req_q_data2_o [LV] => VAR 0x4cedc70 <e2308037#> {cb116cf} @dt=0x4cedb50@(w91)ref  d_rn_req_q_data2_o OUTPUT PORT
```

And it seems like since the very beginning the ASSIGN node has the two VARREFS. Can you suggest any approach to get more info about this?

Thanks

wsnyder, 2023-05-04:
That is the right sort of data, but those are normal assignments not something that would make the problem you describe.  We need both VARREFS to be SystemC which is marked by "[SC]" that is an ASSIGN with underneath two "VARREF ... -> VAR .. [SC]".  I suspect you will find somewhere in a later .tree file this appears.

Sustrak, 2023-05-05:
Indeed they appear on later tree files. Here are the firs appearance with some context:
```
obj_dir/Vlc_th_026_dfg-optimize.tree-    1:2: ASSIGNW 0x1a6fddd0 <e6477942#> {cb722bp} @dt=0x5162b50@(w91)ref
obj_dir/Vlc_th_026_dfg-optimize.tree-    1:2:1: VARREF 0x1a6fdbb0 <e6477939#> {cb68cf} @dt=0x514c450@(w91)ref  d_rn_req_q_data_o [RV] <- VAR 0x514c570 <e2307794> {cb68cf} u3=0x1 @dt=0x514c450@(w91)ref  d_rn_req_q_data_o OUTPUT PORT 
obj_dir/Vlc_th_026_dfg-optimize.tree:    1:2:2: VARREF 0x1a6fdcc0 <e6477940#> {cb116cf} @dt=0x5162b50@(w91)ref  d_rn_req_q_data2_o [LV] => VAR 0x5162c70 <e2308037> {cb116cf} @dt=0x5162b50@(w91)ref  d_rn_req_q_data2_o OUTPUT PORT 

obj_dir/Vlc_th_027_inline.tree-    1:2: ASSIGNW 0x1a6fddd0 <e6477942> {cb722bp} @dt=0x5162b50@(w91)ref
obj_dir/Vlc_th_027_inline.tree-    1:2:1: VARREF 0x1a6fdbb0 <e6477939> {cb68cf} @dt=0x514c450@(w91)ref  __Vcellout__lc_th__d_rn_req_q_data_o [RV] <- VAR 0x15d87420 <e5142178> {cb68cf} @dt=0x514c450@(w91)ref  __Vcellout__lc_th__d_rn_req_q_data_o MODULETEMP
obj_dir/Vlc_th_027_inline.tree:    1:2:2: VARREF 0x1a6fdcc0 <e6477940> {cb116cf} @dt=0x5162b50@(w91)ref  __Vcellout__lc_th__d_rn_req_q_data2_o [LV] => VAR 0x15d7b290 <e5142808> {cb116cf} @dt=0x5162b50@(w91)ref  __Vcellout__lc_th__d_rn_req_q_data2_o MODULETEMP

obj_dir/Vlc_th_029_dfg-optimize.tree-    1:2: ASSIGNALIAS 0x1d570920 <e7202342> {cb116cf} @dt=0x5162b50@(w91)ref
obj_dir/Vlc_th_029_dfg-optimize.tree-    1:2:1: VARREF 0x20357730 <e7843557#> {cb116cf} @dt=0x514c450@(w91)ref  d_rn_req_q_data_o [RV] <- VAR 0x13494f30 <e4278379> {cb68cf} @dt=0x514c450@(w91)ref  d_rn_req_q_data_o [SC] [PO] OUTPUT [P] PORT 
obj_dir/Vlc_th_029_dfg-optimize.tree:    1:2:2: VARREF 0x1dae10e0 <e7202340> {cb116cf} @dt=0x5162b50@(w91)ref  d_rn_req_q_data2_o [LV] => VAR 0x5162c70 <e2308037> {cb116cf} u3=0x1 @dt=0x5162b50@(w91)ref  lc_th__DOT__d_rn_req_q_data2_o PORT 

obj_dir/Vlc_th_029_dfg-optimize.tree-    1:2: ASSIGNW 0x1fbb6780 <e7729748#> {cb116cf} @dt=0x5162b50@(w91)ref
obj_dir/Vlc_th_029_dfg-optimize.tree-    1:2:1: VARREF 0x1fbb6560 <e7729745#> {cb116cf} @dt=0x514c450@(w91)ref  d_rn_req_q_data_o [RV] <- VAR 0x13494f30 <e4278379> {cb68cf} @dt=0x514c450@(w91)ref  d_rn_req_q_data_o [SC] [PO] OUTPUT [P] PORT 
obj_dir/Vlc_th_029_dfg-optimize.tree:    1:2:2: VARREF 0x1fbb6670 <e7729746#> {cb116cf} @dt=0x5162b50@(w91)ref  d_rn_req_q_data2_o [LV] => VAR 0x1349ecb0 <e4278649> {cb116cf} @dt=0x5162b50@(w91)ref  d_rn_req_q_data2_o [SC] [PO] OUTPUT [P] PORT 
```
Please also find attached all the appearances of `d_rn_req_q_data2_o` in the tree files.
[verilator_tree.txt](https://github.com/verilator/verilator/files/11404859/verilator_tree.txt)

Thanks for your help

wsnyder, 2023-05-05:
@gezalore perhaps Dfa optimizer isn't disabling substitutions when a variable has varp()->isSc() set?

closed 2023-05-27 by wsnyder