verilator/verilator #4140: -CFLAGS -O0 has no effect on g++

PeterMonsson, 2023-04-21:
My goal is to bring down incremental compile + run times for simple unit test to 1 second. Currently I am at 5 seconds. 

I use ccache which helps a lot, but I see that the last g++ command takes about 4 seconds. It has the -Os option by default. Manually replacing this option with -O0 speeds up compilation with 1-2 seconds. Unfortunately passing -O0 to verilator with -CFLAGS has no effect. 

I would like to fix this with some guidance. Where should I look? 

Are there other speedups that I am overlooking?

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

What 'verilator --version' are you using?  Did you try it with the git master version?

Master

What OS and distribution are you using?

centOS/Linux 

May we assist you in trying to fix this in Verilator yourself?

Yes

wsnyder, 2023-04-23:
Verilator put OPT_FAST/OPT_SLOW/OPT_GLOBAL after CPPFLAGS in verilated.mk.in.  Fixed.

closed 2023-04-23 by wsnyder

PeterMonsson, 2023-04-26:
Thank you! This resulted in the following turn-around-time improvements for me:

With -O0: 3.6s
Without -O0: 5.3s