verilator/verilator #4141: sprintf is depreciated in favor on a more safe snprintf on macOS 13

allexoll, 2023-04-22:
Currently compiling a verilated project gives depreciated warnings about sprintf, in favor of snprintf

```
In file included from /opt/homebrew/Cellar/verilator/5.006/share/verilator/include/verilated_fst_c.cpp:37:
/opt/homebrew/Cellar/verilator/5.006/share/verilator/include/gtkwave/fstapi.c:5637:42: warning: 'sprintf' is deprecated: This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead. [-Wdeprecated-declarations]
                                wx_len = sprintf(wx_buf, "#%" PRIu64 "\n", time_table[i]);
```

```
# installed trough homebrew
verilator --version
Verilator 5.006 2023-01-22 rev UNKNOWN.REV
```

using macOS 13.3.1 (22E261)

May we assist you in trying to fix this in Verilator yourself?
I can try to make a PR, if there is no other reason than "it has not been done yet" such as "X system does not support it"

wsnyder, 2023-04-22:
That is code from gtkwave, https://github.com/gtkwave/gtkwave, please file a bug there, then once fixed in their repository let us know and we'll update our submodule pull from their code.

allexoll, 2023-04-22:
thanks, i'll head over that way!

wsnyder, 2023-05-03:
No need to keep this open here, when next release I'll automatically see the gtkwave upstream change and merge.  Or, if after gtkwave changes if you want it in Verilator right away, feel free to post to this closed issue and I'll pull from upstream.

closed 2023-05-03 by wsnyder

allexoll, 2023-06-03:
FIY this has been done over at gtkwave in [this commit](https://github.com/gtkwave/gtkwave/commit/048a5c1a5d5b9235720aec4dae852dd25324702c), so it will be included next time you "pull" from gtkwave